StanfordAHA/garnet #12: Module prototype

phanrahan, 2018-07-14:
We should make a module as clean as possible,.

For example, in cb

How about

cb.py -functional model
cb_magma - magma version
cb_genesis2 - genesis2 wrapper version

Maybe we can merge the main program for the genesis2 version into cb_genesis2?

leonardt, 2018-07-14:
`cb.vp` is currently in test_cb, but I think it makes sense to move it to the `cb` directory.

leonardt, 2018-07-14:
I think merging the main and wrapper makes sense. Just guard the main/argparse functionality with "if __name__ == "__main__":"

leonardt, 2018-07-14:
I updated my mem PR to match the proposed organization, here's the consolidated file: https://github.com/rsetaluri/magma_cgra/blob/4654a688b432837806a5501b34385319cc9b1093/mem/mem_genesis2.py

I think this organization makes sense.

rsetaluri, 2018-07-16:
Agreed. I've updated cb to this convention as well (see bfe4c63). Closing.

closed 2018-07-16 by rsetaluri