StanfordAHA/garnet #14: tests

phanrahan, 2018-07-14:
The file test_cb/test_regressions.py should be generalized to take any module (e.g. cb) and generate the tests. The boilerplate should be moved into the test infrastructure. 

We should strive to build a clean test infrastructure so that it is easy for developers to run common tests on their modules.

leonardt, 2018-07-14:
Makes sense, here's a proposed initial interface.
`test(genesis_circuit, functional_model)`
We can generate the tests based on the functional model, but there's one subtle detail with the configuration logic. There's a sequence of pokes that have to occur before each test to setup the circuit, and the functional_model configuration method needs to be called. We could abstract this into a reusable test component for configurable modules, so maybe something like
`test_configurable(genesis_circuit, function_model, config_vectors)` and the user has to provide the configuration sequence using the Tester.poke model.

phanrahan, 2018-07-14:
Good catch. Yes, we should write a special version of the test infrastructure that correctly handles configuration.

Maybe something like 

```
test(configuration, genesis_circuit, functional_model)
```

phanrahan, 2018-07-16:
I propose we use uppercase for acronyms. So connect_box would become CB (not Cb). Similarly, FPGA should not be spelled Fpga.