StanfordAHA/garnet #22: Repo build issues

rsetaluri, 2018-07-16:
From @kong0329:

> Hi Raj,
I have questions about magma_cgra.

1. When I run `python cb/cb_wrapper_main.py test_cb/cb.vp` command, an error occurs.
```Traceback (most recent call last):
  File "cb/cb_wrapper_main.py", line 6, in <module>
    from cb.cb_wrapper import define_cb_wrapper
ModuleNotFoundError: No module named 'cb.cb_wrapper'; 'cb' is not a package```
I suspected that same name of package name(cb) and module name(cb.py) so removed cb.py and tried again.
Now I get this error.
```Traceback (most recent call last):
  File "cb/cb_wrapper_main.py", line 6, in <module>
    from cb.cb_wrapper import define_cb_wrapper
ModuleNotFoundError: No module named 'cb'```
Could you let me know how to resolve it?

2. This is to make sure what I understood about magma_cgra is correct.
<Let's say we design new module X>
a. Implement X.vp (Genesis generator file)
b. Implement wrapper. X_wrapper.py and X_wrapper_main.py
c. Implement magma file. X.py
d. Implement functional file. X_functional.py
e. Implement testbench. X_tb.v
f1. test_wrapper - test if wrapper works.
f2. test_functional - test if functional model is correct
f3. test_regression - test if generator magma and genesis behave same. test both behave as functional model.

Is this correct?
In the future, can we just skip a, b, f1 if we only stick to magma, not genesis?

3. Lastly, the point of magma is to increase the abstraction layer to python(magma), right?
If so, how is testbench file X_tb.v generated automatically according to parameters?
If we still have to implement testbench file in verilog, it would be inefficient.

Thanks for having great system Raj.

rsetaluri, 2018-07-16:
From @rsetaluri:

Hi Taeyoung, good questions

raj [5:19 PM]
1) you do not need to change around the file names for this. checkout the Usage section of the readme. particularly

```pip install -r requirements.txt  # install python dependencies
pytest                           # install with pip install pytest``` 

actually, you will also need to do `pip install -e .` to run not just the tests but also `cb_wrapper_main.py`
Let me know if that works

2) Yes this is mostly correct. We are actually changing around some of the file names. This issue — https://github.com/rsetaluri/magma_cgra/issues/12 -- has what we want to do going forward regarding naming conventions. See the `mem` directory to see an example fo this
Yes, without Genesis we do not need those steps. But we do not think that we will move away from verilog entirely soon (edited)
3) Yes that is the intended use of magma. The testbench file `X_tb.v` is not necessarily a fundamental piece of this and would not generally be automatically generated. In this example we’ve included it just to show that you can also hook up with other testing paradigms. Especially hardware people who are already familiar with system verilog test benches can also use our system.
However, we are working to create a layer of abstraction on top of testing — you are right, it is not convenient to write out tests by hand. We are working on a high level system which you can pass in a functional model or test vectors and then automatically generate a set of tests which can be run against genesis version, magma version, etc.
For now you do not need to write a verilog based test bench (you can if you would like however). For now we are focused on adding functional models + wrappers + tests following the format of the cb module. We will slowly add magma versions
Does this make sense?

rsetaluri, 2018-07-16:
From @kong0329:

Hi Raj,
Thanks for your answers. They really helped a lot.
I still have some follow-up questions, so I appreciate if you could answer them.

2) This is clear. Thanks a lot.
3) If you slowly add magma, then until you have magma, the main goal of this system is to have functional model in python and easily test it with high level python, right?

1) For this, I am still facing some issues, and I think all these issues are just setup issues, which is annoying haha.
• `pytest test_cb/test_regression.py`
When I run this command, test fails in kiwi.
I found that it is related to coreir library path.
Even though I installed coreir package by running `python setup.py install`, running `pytest` couldn't find the library path, so I had to run below commands.

```
wget https://github.com/rdaly525/coreir/releases/download/v0.0.12/coreir.tar.gz
mkdir coreir_release;
tar -xf coreir.tar.gz -C coreir_release --strip-components 1;
export PATH=~/coreir_release/bin:$PATH;
export LD_LIBRARY_PATH=~/coreir_release/lib:$LD_LIBRARY_PATH;
```

Is there another way other than adding coreir library path to `.bashrc`?

• `pytest test_cb/test_regression_ncsim.py`
When I run this command, test aborted with below output.

```
test_cb/test_regression_ncsim.py Aborted (core dumped)
```

I further investigated it, and found `compile_to_verilog` function fails to open cached json file, but I am not sure.
Since ncsim.py is not tested in travis machine, I wonder whether it is working in any local machine.
Could you let me know how to solve it?

• `python cb/cb_wrapper_main.py test_cb/cb.vp`
This command doesn't work for me. Could you check whether above command works?
If `cb.py` file is in `cb` folder, `cb_wrapper_main.py` regards cb in `import cb.cb_wrapper` as module, not package.

Many thanks, Raj!

Thanks

leonardt, 2018-07-16:
@kong0329 I'm assuming you're running on kiwi.
I was able to reproduce the issue with `test_cb/test_regression_ncsim.py Aborted (core dumped)`, it's because we're using `/tmp` on a shared machine and the file it tried to create already existed, but was owned by a different user so we didn't have permission to edit it.

The commit in the branch `compile-with-tempdir` should fix it, you can try it out with `git checkout compile-with-tempdir`, or wait until it's merged into mater.

leonardt, 2018-07-16:
@kong0329 for `python cb/cb_wrapper_main.py test_cb/cb.vp`, if you're on the latest master this won't work since the files have been renamed/reorganized. Can you try `python cb/cb_genesis2.py cb/genesis/cb.vp`? Also, if you can include the error log, that would be helpful.

leonardt, 2018-07-16:
You can install `coreir` without editing your path by installing it directly into your system path, see 
https://github.com/rsetaluri/magma_cgra/pull/26/files for example instructions. The release comes with a makefile that allows you to use `make install`

rsetaluri, 2018-07-16:
Closing. See pr #25 for @kong0329 's approval.

closed 2018-07-16 by rsetaluri