StanfordAHA/garnet #63: Support wrapping multiple (different) generate calls to the same genesis module

rsetaluri, 2018-07-31:
Say we have a generator defined as a wrapper around genesis called `define_foo_wrapper()`.

Then:

```
def define_bar(...):
    class _Bar(m.Circuit):
        IO = [...]

        @classmethod
        def definition(io):
            ...
            foo_inst0 = define_foo_wrapper(x)()
            foo_inst1 = define_foo_wrapper(y)()
            ...
```

will not produce correct verilog. Basically, the only one `foo` module will be imported. I think the underlying issue is that the circuits returned by `define_foo_wrapper(x)` and `define_foo_wrapper(y)` will have the same **name**.

One solution is to change the name of the circuit returned from `DefineFromVerilog()`. (or pass in a name to that function).

Maybe a good time to look into linking properly @leonardt @rdaly525

leonardt, 2018-08-02:
Is this because genesis doesn't generate a new name based on different parameters?

rsetaluri, 2018-08-02:
Yes -- to be more precise, it **does** generate new names based on parameters (e.g. `foo_unq1.v`, `foo_unq2.v`). But since the top level module is only ever passed 1-set of parameters (by virtue of being the top), genesis doesn't need to do anything special, so we just get `top.v` with `module top(...)` inside.

It wouldn't be feasible for Genesis to generate separate top level names *across* runs of Genesis. There's probably a proper thing to do here. A hack is to have the wrapper instance a new module around the one returned from `DefineFromVerilog` where we just pass through ports, and give a custom name. Or add an option to add a name in `DefineFromVerilog`.

I think the general question here might be: how would you import multiple modules with the same name in their respective namespace. A python analog is like if you have two packages `package1.util` and `package2.util`,  you can't import them both like `from package{x} import util` without overwriting (so we leverage the namespaces there).