StanfordAHA/garnet #84: CB regressions and test_config_reg failing on Travis

alexcarsello, 2018-08-15:
I just pushed to my gc_functional_model branch, and I'm getting unrelated test failures.

I also kicked off another Travis run for the master branch and I'm seeing the same failures.

[https://travis-ci.org/StanfordAHA/garnet/builds/415246081](url)

rsetaluri, 2018-08-15:
Good catch. I think these are errors due to version mismatches in magma/mantle.

@leonardt can we freeze the external versions we refer to in `garnet` to avoid this? In the meantime not sure what the specific issue here is...

leonardt, 2018-08-15:
It's a little harder since we haven't released PYPI packages for magma/mantle yet. We could try to fix a commit hash in https://github.com/StanfordAHA/garnet/blob/master/requirements.txt

I'm going to try to get a PYPI release of magma/mantle for the facebook hackathon so maybe we can just wait for that. Then we should be able to pin the version in setup.py

leonardt, 2018-08-15:
There's an issue with commonlib.muxn from coreir that uses the `join` name for an instance. This is a reserved verilog keyword that's not supported by verilator, so we get
```
Unsupported: Verilog 1995 reserved word not implemented: join
```

I'm submitting a patch to coreir to update the muxn implementation not to use join.

leonardt, 2018-08-16:
This https://github.com/StanfordAHA/garnet/pull/85 along with this https://github.com/StanfordAHA/garnet/pull/86 resolves the issue.

You can rebase on `update-config-register` to get a working copy, or wait for the merge to master.

rsetaluri, 2018-08-16:
@alexcarsello things should be merged. Rebase and check that things still work.

alexcarsello, 2018-08-16:
Looks good, thanks.

closed 2018-08-16 by alexcarsello