StanfordAHA/garnet #99: Most wrapper tests failing

alexcarsello, 2018-09-07:
[https://travis-ci.org/StanfordAHA/garnet/builds/425093501](url)

Looks like the interface that Magma outputs has changed, which is causing most of the wrapper tests that check the output to fail on Travis.

rsetaluri, 2018-09-07:
This change caused that: https://github.com/phanrahan/magma/commit/18943be3fcba297b97ba2ce401eb03d8d1159a8f

will fix this by changing the expectations in the tests

rsetaluri, 2018-09-08:
@alexcarsello can you check that the problem is fixed now?

alexcarsello, 2018-09-10:
Yes it looks like it's fixed. Just need to update Magma on Kiwi so that it matches what's running on Travis.

alexcarsello, 2018-09-10:
@leonardt I was actually just about to ask you about this. I think writing up a doc would be a good idea since this has come up a few times before.

leonardt, 2018-09-10:
@alexcarsello instructions added here https://github.com/StanfordAHA/garnet/wiki/Kiwi-Environment#updating-python-dependencies

alexcarsello, 2018-09-10:
Tried to go through the instructions, and I think the address for the "git clone" is bad... should be https://github.com/phanrahan/magma.git

closed 2018-09-10 by rsetaluri