StanfordAHA/garnet #102: Figure out why GC coverage decreases

rsetaluri, 2018-09-11:
See: https://coveralls.io/builds/18924947

The file where decreased coverage is reported is untouched. Unsure why it is reporting decreased coverage. Maybe some thing is stale.

leonardt, 2018-09-12:
So looking at https://github.com/StanfordAHA/garnet/pull/101, I see 
```
Change from base Build 305: | -0.1%
```

For Build 305, if we look at https://coveralls.io/builds/18922701/source?filename=global_controller/global_controller.py#L124
and Build 310 (the one referenced in the top comment), https://coveralls.io/builds/18924947/source?filename=global_controller/global_controller.py#L124

We see that line 124 isn't covered. Is there some non-determinism in the test case that could cause this line to be occasionally uncovered? E.g. maybe there's a case when no mask bit is set.

@alexcarsello do you know how that mask is generated in the test case?

rsetaluri, 2018-09-13:
I notice a lot of random's in that the GC tests. @alexcarsello maybe you can do a pass and make it pseudo-constrained-random (maybe just hard code some special cases) to ensure coverage? Not very urgent. In the meantime we can add `#nocover` to them.

alexcarsello, 2018-09-13:
Yes there is some non-determinism there, so it's possible that line 124 could occasionally be uncovered. I'll fix this.