StanfordAHA/garnet #115: Simple CGRA issues

rsetaluri, 2018-09-20:
[ ] Add config enable logic, anded with R/W signal
[ ] Switch to using `MuxWithDefault` rather than `Mux` for `config_read_data` logic in simple_cb/sb
[ ] Figure out `cannot collect test class 'Tester' because it has a __init__ constructor` warning in pytest

rsetaluri, 2018-09-20:
`cannot collect test class 'Tester' because it has a __init__ constructor` happens if we have a class named `Test*` because pytest discovers this class as a test class (and if it has an `__init__` method, it skips it). `from fault import Tester` brings in the `Tester` class into this scope. Instead we should opt for `import fault` and `fault.Tester...`

leonardt, 2018-09-20:
You could try this change (https://github.com/leonardt/fault/commit/520101f3db6552ab7dcb33aae25a8952046d6379) on the `test-false` branch of fault

From what I read, that's supposed to tell pytest to skip the class when doing test discovery. If it works we can merge it into master and release

rsetaluri, 2018-09-20:
Confirmed that it works.