StanfordAHA/garnet #116: Proposal for testing paradigm

rsetaluri, 2018-09-20:
I've been thinking about the whole model/driver/monitor, transaction level modeling stuff. The high level idea is that a model has high-level functions, a driver lowers these functions to circuit inputs, and a monitor lifts circuit outputs to high-level outputs. An important thing to note is that a model should be 1-to-1 with a functional specification, where as there can be many circuits that adhere to that spec (i.e. there exists a driver-monitor pair that makes the circuit and spec compatible). For example, we could simply rename the ports of a circuit without really changing the meaning of the circuit.

It seems like when designing a module there should be one "model" and then a driver-monitor pair for each circuit which should adhere to the model, and the *circuit designer* is responsible for writing the driver and monitor. In particular, the driver and monitor should have the same set of functions, with the driver issuing "pokes" based on function inputs, and the monitor issuing "expects" based on model outputs.

Here is a prototype of this for a simple counter circuit. I think the general work flow is nice, but there's a lot of repeated stuff (primarily the class interfaces/function names -- though the bodies of each function are disjoint). Also, it is not easy to do things that don't fit into the simple driver-monitor framework. For example, you may want to do edge-case checks, contra-positive checks, etc. That doesn't fit easily here.

```python
from bit_vector import BitVector
import fault
import magma
import mantle

class CounterModel:
    def __init__(self, width):
        self.width = width

        self.O = 0

    def step(self):
        self.O += 1

class CounterDriver:
    def __init__(self, width, tester):
        self.width = width
        self.tester = tester
        self.circuit = self.tester.circuit

    def step(self):
        self.tester.step(2)

class CounterMonitor:
    def __init__(self, width, tester, model):
        self.width = width
        self.tester = tester
        self.circuit = self.tester.circuit
        self.model = model

    def step(self):
        self.tester.expect(self.circuit.O, self.model.O)

class CounterTester:
    def __init__(self, model, driver, monitor):
        assert monitor.tester is driver.tester
        assert monitor.model is model

        self.model = model
        self.driver = driver
        self.monitor = monitor

    def step(self):
        self.model.step()
        self.driver.step()
        self.monitor.step()

if __name__ == "__main__":
    width = 4

    model = CounterModel(width)
    circuit = mantle.DefineCounter(width)
    tester = fault.Tester(circuit, circuit.CLK)
    driver = CounterDriver(16, tester)
    monitor = CounterMonitor(16, tester, model)

    full_tester = CounterTester(model, driver, monitor)
    for i in range(10):
        full_tester.step()
    tester.compile_and_run(target="verilator", directory="/tmp/", magma_output="coreir-verilog")
```

rsetaluri, 2018-09-20:
Here is another approach that is more concise. It squashes the driver and monitor into one, and then uses the `tester_function` decorator to automatically call the same function with the same arguments to the model. @leonardt I know you've done some work on the functional tester paradigm in fault/garnet, I think this is a next iteration of that.

```python
from bit_vector import BitVector
import fault
import magma
import mantle

def tester_function(fn):
    def _fn(*args, **kwargs):
        this = args[0]
        model = getattr(this, "model")
        fn_name = fn.__name__
        getattr(model, fn_name)(*args[1:], **kwargs)
        fn(*args, **kwargs)
    return _fn

class CounterModel:
    def __init__(self, width):
        self.width = width

        self.O = 0

    def step(self):
        self.O += 1

class FunctionalTester:
    def __init__(self, model, tester):
        self.model = model
        self.tester = tester
        self.circuit = self.tester.circuit

class CounterTester(FunctionalTester):
    def __init__(self, model, tester, width):
        super().__init__(model, tester)

        self.width = width

    @tester_function
    def step(self):
        self.tester.step(2)
        self.tester.expect(self.circuit.O, self.model.O)

if __name__ == "__main__":
    width = 4

    model = CounterModel(width)
    circuit = mantle.DefineCounter(width)
    tester = fault.Tester(circuit, circuit.CLK)

    full_tester = CounterTester(model, tester, width)
    for i in range(1):
        full_tester.step()
    tester.compile_and_run(target="verilator", directory="/tmp/", magma_output="coreir-verilog")
```

leonardt, 2018-09-21:
This all looks good to me.

One thing to think about is decoupling the monitor from the tester. I think the proposed pattern is useful, but one could also want to write a monitor that just uses a peek abstraction, for example:
```
class CounterMonitor:
    def step(self):
        self.tester.expect(self.circuit.O, self.circuit.I + 1)
```

note that this doesn't work because the expression for the expected output is computed in Python (but it references a test variable). More generally, it could be viable to compile expressions/statements from the test code into the harness (or execute them directly in an interactive mode with the actual values) which would allow one to write more complex monitors that are functions of current circuit values (or state if we add a mechanism for lifting state).

I think the current monitor you proposed as lifting the raw ports to the functional model outputs is a good application of the pattern to the functional model testing pattern, but I think what I'm trying to say is this is just one possible application of the monitor pattern.

In summary, I think what you're proposing is a way to apply the driver and monitor pattern from UVM to our functional model based tests. I think what you're proposing sounds like a good way to do it. I think my above note is just mentioning possibilities for the monitor abstraction in a more general pattern, but isn't specifically related to functional models (e.g. writing monitors that just monitor the state of the circuit).

I think the idea of folding the monitor and driver into a single component makes sense in the case that the actions of each (poke and expect) can be done in sequential order (reads well in the code). I'd imagine it's more useful to break them up if there's some variable timing behavior. For example, the driver could send some valid data in, step the circuit for a number of cycles. Then, we can either expect the output (in the combined case), or the monitor could just check an output valid bit. Now that I think about this, we need to consider how the functional model, driver, and monitor relate in terms of timing semantics. Basically, the driver could be responsible for stepping the circuit the amount of time needed for the output to be ready, or the monitor could do this. Currently, the functional model has no notion of time, so it's up to the *circuit designer* to manage this mapping. I think if we consider more complex cases with variable input/output streams, it may be useful to separate the driver/monitor, but in the cases we've seen so far, it seems *okay* to keep merge them into the same component (and convenient for the test author).

Small note:
The repeated code could between the monitor/driver classes could be shared by some parent class (UVM calls these testing components, so we could have a `CounterComponent` which defines shared attributes).