StanfordAHA/garnet #129: Cannot run tapeout branch and partial solutions

Kuree, 2018-10-27:
I'm trying to figure out how to run `garnet`. It seems like `tapeout` branch, rather than `master` branch, has the top module that I can use to generate PnR info. 

Here is what I tried:
`$ pytest .`
```
ImportError while importing test module '/home/keyi/garnet/test_top/test_top_magma.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
test_top/test_top_magma.py:2: in <module>
    from top.top_magma import CGRA
top/top_magma.py:3: in <module>
    import generator.generator as generator
generator/generator.py:2: in <module>
    from ordered_set import OrderedSet
E   ModuleNotFoundError: No module named 'ordered_set
```

It seems like `ordered-set` was not installed (not in the `requirements.txt`). So I did `pip install ordered-set` and it seems to pass this check.

Now try `pytest .` again, which gives me 
```
ImportError while importing test module '/home/keyi/garnet/test_common/test_config_register.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
test_common/test_config_register.py:4: in <module>
    from magma.bit_vector import BitVector
E   ModuleNotFoundError: No module named 'magma.bit_vector'
```

Well after digging around it seems like just a naming error. So I did the following changes:
```diff
$ git diff test_common/test_config_register.py
diff --git a/test_common/test_config_register.py b/test_common/test_config_register.py
index 8e79443..b9bb64e 100644
--- a/test_common/test_config_register.py
+++ b/test_common/test_config_register.py
@@ -1,7 +1,7 @@
 import os
 import filecmp
 import magma as m
-from magma.bit_vector import BitVector
+from bit_vector import BitVector
 from magma.simulator.coreir_simulator import CoreIRSimulator
 from common.config_register import define_config_register
 from common.util import check_files_equal
```

That seems to bring out pytest. However, somehow `test_simple_pe/build/pe.json` got modified during the process and thus failed the test. 
```
parse error - unexpected end of input; expected '}'

ERROR: No top in file :test_simple_pe/build/pe.json
```
```diff
$ git diff test_simple_pe/build/pe.json
diff --git a/test_simple_pe/build/pe.json b/test_simple_pe/build/pe.json
index 1c0c7db..fad51ba 100644
--- a/test_simple_pe/build/pe.json
+++ b/test_simple_pe/build/pe.json
@@ -144,4 +144,3 @@
     }
   }
 }
-}
\ No newline at end of file
```

I don't know garnet well enough to figure out which test modifies that file. I will setup inode watcher to monitor the filesystem changes to see which part goes wrong.

leonardt, 2018-10-30:
BitVector issue was fixed by https://github.com/StanfordAHA/garnet/commit/8d10f73c9e7e10b612e74b5c5c7196a8d815b71b, looks like what you have checked out is out of date with master.

The pe.json issue was also fixed in the same PR as https://github.com/StanfordAHA/garnet/pull/124/files so I think your branch hasn't been updated since that latest PR into master. You should rebase that branch onto master.

The `ordered_set` issue looks like it should just be added to `setup.py` or `requirements.txt`, with `setup.py` being preferred. I think that package is only a dependency for that branch, since I don't see a reference to that in master.

Kuree, 2018-10-30:
@leonardt Thanks! I will close this issue for now.

@alexcarsello Can you rebase `tapeout` branch onto `master`?

closed 2018-10-30 by Kuree