StanfordAHA/garnet #152: test_memory_core fails with verilator 4.008

leonardt, 2019-01-28:
According to the travis log, it is using verilator 3.856 (https://api.travis-ci.com/v3/job/173092884/log.txt):

    Get:5 http://us-east-1.ec2.archive.ubuntu.com/ubuntu trusty/universe amd64 verilator amd64 3.856-1 [2,180 kB]

Confirmed on MacOS and CentOS using verilator 4.008 causes test_memory_core/test_memory_core.py to fail.

MacOS (homebrew) log

```
======================================================================================== test session starts =========================================================================================
platform darwin -- Python 3.7.0, pytest-4.1.1, py-1.7.0, pluggy-0.8.1
rootdir: /Users/lenny/repos/garnet, inifile: setup.cfg
collected 2 items

test_memory_core/test_memory_core_genesis2.py .F                                                                                                                                               [100%]

============================================================================================== FAILURES ==============================================================================================
__________________________________________________________________________________________ test_sram_basic ___________________________________________________________________________________________

    def test_sram_basic():
        generator = memory_core_genesis2.memory_core_wrapper.generator(
            param_mapping=memory_core_genesis2.param_mapping)
        Mem = generator()  # Using default params
        for genesis_verilog in glob.glob("genesis_verif/*.v"):
            shutil.copy(genesis_verilog, "test_memory_core/build")

        # FIXME: HACK from old CGRA, copy sram stub
        shutil.copy("test_memory_core/sram_stub.v",
                    "test_memory_core/build/sram_512w_16b.v")

        # Setup functiona model
        DATA_DEPTH = 1024
        DATA_WIDTH = 16
        MemFunctionalModel = gen_memory_core(DATA_WIDTH, DATA_DEPTH)
        mem_functional_model_inst = MemFunctionalModel()

        tester = MemoryCoreTester(Mem, clock=Mem.clk_in,
                                  functional_model=mem_functional_model_inst)
        tester.zero_inputs()
        tester.expect_any_outputs()

        tester.eval()

        tester.poke(Mem.clk_en, 1)

        tester.reset()

        mode = Mode.SRAM
        tile_enable = 1
        depth = 8
        config_data = mode.value | (tile_enable << 2) | (depth << 3)
        config_addr = BitVector(0, 32)
        tester.configure(config_addr, BitVector(config_data, 32))
        num_writes = 20
        memory_size = 1024

        def get_fresh_addr(reference):
            """
            Convenience function to get an address not already in reference
            """
            addr = random.randint(0, memory_size - 1)
            while addr in reference:
                addr = random.randint(0, memory_size - 1)
            return addr

        addrs = set()
        # Perform a sequence of random writes
        for i in range(num_writes):
            addr = get_fresh_addr(addrs)
            # TODO: Should be parameterized by data_width
[... 366 lines cut ...]
%Warning-UNUSED: input_sr_unq1.v:78: Signal is not driven, nor used: ___genesis_wire_16
%Warning-UNUSED: output_sr_unq1.v:77: Signal is not driven, nor used: ___genesis_wire_16
%Warning-CASEINCOMPLETE: memory_core.v:199: Case values incompletely covered (example pattern 0x1)
%Warning-CASEINCOMPLETE: linebuffer_control_unq1.v:222: Case values incompletely covered (example pattern 0x2)
%Warning-COMBDLY: input_sr_unq1.v:120: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
%Warning-COMBDLY: *** See the manual before disabling this,
%Warning-COMBDLY: else you may end up with different sim results.
%Warning-COMBDLY: input_sr_unq1.v:132: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
%Warning-COMBDLY: input_sr_unq1.v:152: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
%Warning-COMBDLY: output_sr_unq1.v:119: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
%Warning-COMBDLY: output_sr_unq1.v:135: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
%Warning-COMBDLY: output_sr_unq1.v:149: Delayed assignments (<=) in non-clocked (non flop or latch) block; suggest blocking assignments (=).
%Warning-BLKSEQ: sram_512w_16b.v:40: Blocking assignments (=) in sequential (flop or latch) block; suggest delayed assignments (<=).
%Warning-UNOPTFLAT: memory_core.v:147: Signal unoptimizable: Feedback to clock or circular logic: memory_core.config_mem
make: Warning: File `Vmemory_core__ver.d' has modification time 19 s in the future
make: warning:  Clock skew detected.  Your build may be incomplete.

Got      : 0x0
Expected : 0x1df
i        : 1277
Port     : data_out
------------------------------------------------------------------------------------- Captured log call -------------------------------------------------------------------------------------
verilog.py                 221 DEBUG    compiling circuit memory_core
============================================================================ 1 failed, 1 passed in 4.18 seconds =============================================================================
```

leonardt, 2019-01-28:
Test passes using verilator v3.888 on MacOS and CentOS

Kuree, 2019-06-15:
Closed for now since the memory core has also been rewritten. I don't think this issue is relevant any more.

closed 2019-06-15 by Kuree