StanfordAHA/garnet #193: Winter 2019 Integration Hackathon

phanrahan, 2019-03-07:
This issue is to track the tasks that need to be done to achieve the tools and cgra goals for this quarter.

1. Switch to Lassen
2. Generate CGRA (Garnet) with Lassen and new interconnect
3. Automatically build mapper, place-and-route graph, and configuration engine
4. Compile halide program (point-wise operation and 3x3 convolution) using floating point to coreir
5. Push halide coreir through the tools to generate bitstream
6. Full system simulation in verilog of Garnet running the bitstream
7. Evaluate area and power 

Tasks
- [ ] Merge Lassen PRs
- [ ] Generate RTL from Lassen
- [ ] Add configuration registers to Peak
...

leonardt, 2019-03-07:
I added 3 sub tasks for generating RTL
* Converge syntax from magma prototype pe1 and lassen description
  * The prototype lassen pe that I have compiled has some changes to the syntax required to support the m.sequential syntax. I'll either need to build out the frontend support in magma for the current peak/pe1 syntax, or merge my syntax changes into the mainline. I'll work on a PR with the syntax changes so we can discuss whether magma needs to be augmented to support it, or whether it's okay to make the peak/pe1 level changes.
* Add floating point types to magma, operators to mantle
  * This is straightforward since it's identical to the magma/mantle support for UInt/SInt types and the coreir primitives
* Verify functionality of generate RTL (coreir sim, verilog sim w/ designware prims)
  * I believe the coreir simulator has support for the floating point types, so we can run the design in simulation to verify it against the executable model. Once that is verified, we can lower the design to using the designware primitives and test them using ncsim or vcs.

jeffsetter, 2019-03-09:
Below are the Halide floating point applications (fp_conv_3_3 and fp_pointwise).

One issue I see with the designs below are the constants are instantiated as regular coreir.const, and perhaps there should be float.const instead that I use. Apart of that, we should be able to get started on these designs.

[image]
[fp_conv_3_3_design_top.txt](https://github.com/StanfordAHA/garnet/files/2948787/fp_conv_3_3_design_top.txt)
[fp_pointwise_design_top.txt](https://github.com/StanfordAHA/garnet/files/2948788/fp_pointwise_design_top.txt)

rdaly525, 2019-03-18:
Right now there is PR on lassen (https://github.com/StanfordAHA/lassen/pull/7) which can successfully discover rewrite rules and map a basic application.

phanrahan, 2019-03-18:
Great!

Is the task list up to date?

Kuree, 2019-03-18:
Can someone post a mapped JSON file so that I can work on it?

rdaly525, 2019-03-18:
[_mapped_add4.txt](https://github.com/StanfordAHA/garnet/files/2980143/_mapped_add4.txt)

jeffsetter, 2019-03-18:
Can you also map the pointwise app above?

Kuree, 2019-03-18:
I can only simulate a 2x2 grid right now. @rdaly525 Can you map pointwise? How did you get this add4 from Halide?

Kuree, 2019-03-18:
@rdaly525 As far as I know, there is no `clk_en` connection box. Is it intentional to have PE clock-gated from application? I thought you can only do that through global controller.

```JSON
["add00$inst.clk_en","add00$c10051.out"],
```

If that is the case, I cannot PnR this application until the `clk_en` port is properly created in Garnet. @rsetaluri @phanrahan Do you know the plan about `clk_en`?

rdaly525, 2019-03-18:
From the Peak perspective, clk_en is simply treated as another input port since it is listed as an input port. It is up to the lassen architect (is there someone specifically in charge of this architecture? @alexcarsello?) to describe this correctly. Also there was a slight hack to get the automatic mapper to run faster (1024x) by assuming that the register was in bypass mode. 

The add4 was just a simple coreir file that I use for testing coreir. It was not created by halide.

Kuree, 2019-03-18:
So when is pointwise going to be mapped? I can some front-end hacks to remove all the unnecessary nets before feeding into the PnR backend if mapping pointwise is going to take some time.

Is `clk_en` a user defined port to Peak? I thought Peak only takes in a very high-level functional descriptions. Maybe I'm wrong?

Kuree, 2019-03-18:
[_mapped_add4.route.txt](https://github.com/StanfordAHA/garnet/files/2980877/_mapped_add4.route.txt)
[_mapped_add4.place.txt](https://github.com/StanfordAHA/garnet/files/2980878/_mapped_add4.place.txt)

Did some changes in the cgra parameters
```diff
--- a/tests/test_interconnect/test_interconnect_cgra.py
+++ b/tests/test_interconnect/test_interconnect_cgra.py
@@ -306,9 +306,9 @@ def create_cgra(chip_size: int, add_io: bool = False):
                     or y in range(margin) \
                     or y in range(chip_size - margin, chip_size):
                 if x == margin or y == margin:
-                    core = IO16bit()
-                else:
                     core = IO1bit()
+                else:
+                    core = IO16bit()
             else:
                 core = MemCore(16, 1024) if ((x - margin) % 2 == 1) else \
                     PECore()
```

Also I'm using a 4x4 array (it was 2x2 by default).

I can't simulate the result with old PE since I've never able to get the garnet testing working for the past 2 weeks.

@rsetaluri Can you generate the bitstream from that at least?

phanrahan, 2019-03-28:
@rdaly525 I noticed that you have 3 unchecked tasks: register mapping, floating point, and ROMs. Are these still open tasks?

rdaly525, 2019-03-28:
@rsetaluri, I have a float test in https://github.com/StanfordAHA/lassen/pull/21. You should be able to add the FPAdd and FPMul mappings

rsetaluri, 2019-05-06:
Closing in favor of #227, which is an updated task list

closed 2019-05-06 by rsetaluri