StanfordAHA/garnet #203: Error in README? No "make install" for CoreIR.

steveri, 2019-04-15:
README for garnet ((https://github.com/StanfordAHA/garnet/blob/master/README.md) says to do this:
```
Install CoreIR
$ git clone https://github.com/rdaly525/coreir.git
$ cd coreir
$ make install -j  # Use sudo on linux
```

...but it looks as though coreir/Makefile (https://github.com/rdaly525/coreir/blob/master/Makefile) does not have an "install" target

```
$ make -n install
$ make: *** No rule to make target 'install'.  Stop.
```

...should be an easy fix, yes?

rdaly525, 2019-04-15:
We are now using cmake. The installation instructions are https://github.com/rdaly525/coreir/blob/master/INSTALL.md

Kuree, 2019-04-18:
Garnet only needs python binding of coreir, which should be installed via
```
pip install coreir
```
On linux it should take less than a second to install.

steveri, 2019-04-18:
Okay, thanks everybody, I updated the README to reflect the new (much simpler) procedure...see
https://github.com/StanfordAHA/garnet/blob/master/README.md for the changes...

closed 2019-04-18 by steveri