StanfordAHA/garnet #215: Unnecessary uniquification of new PE tiles

alexcarsello, 2019-04-27:
In the latest garnet.py, the PE's are being uniquified, even though they have identical verilog module definitions.

leonardt, 2019-04-27:
See https://github.com/phanrahan/magma/pull/385 for a workaround

leonardt, 2019-04-27:
Waiting for feedback on a longer term solution, but that magma branch should generate a single PE tile, the issue was that two definitions of `SB_ID0_5TRACKS_B1_irq_res_p` had the same instances and wiring, but the instances were created in different orders. This means the repr string used for hashing had a different line ordering. The workaround avoids this issue by sorting the instances by name before generating the repr string (but this means the the repr string no longer has definition ordering). We might want to preserve definition order for `repr` (I don't know if anyone cares or uses this currently), if so we can change it to only use the sorted instances logic in the code path specifically for hashing/uniquification.

alexcarsello, 2019-04-27:
Cool, this worked for me. Thanks for the quick response/fix!

closed 2019-05-27 by alexcarsello