StanfordAHA/garnet #216: Very long verilog generation times

alexcarsello, 2019-04-27:
Since the new halide-flow branch was merged in, the time to generate the full 32x16 CGRA has gone up significantly. It's been running for 2 hours now, and it's still not finished.

Kuree, 2019-04-27:
Here is what I tried to speed up the generation without touching the magma backend:
1. I swapped out the `DotDict` with `DotMap` to allow ports to be pickled
2. I change the top level circuit generation in `Interconnect` to the following:
```Python
    def circuit(self):
        children = self.children()
        circuits = {}
        cpu_count = multiprocessing.cpu_count()

        pool = Pool(cpu_count)

        jobs = list(children)

        def map_circuit(child_):
            return child_.circuit()
        job_result = pool.map(map_circuit, jobs)
        count = 0
        for child in children:
            circuits[child] = job_result[count]
            count += 1

        class _Circ(magma.Circuit):
            name = self.name()
            IO = self.decl()

            @classmethod
            def definition(io):
                instances = {}

                def map_inst(child__):
                    kwargs = {}
                    if child__.instance_name:
                        kwargs["name"] = child__.instance_name
                    return circuits[child__](**kwargs)

                job_result_ = pool.map(map_inst, jobs)
                count_ = 0
                for child_ in children:
                    instances[child_] = job_result_[count_]
                    count_ += 1

                instances[self] = io
                for port0, port1 in self.wires:
                    inst0 = instances[port0.owner()]
                    inst1 = instances[port1.owner()]
                    wire0 = port0.get_port(inst0)
                    wire1 = port1.get_port(inst1)
                    magma.wire(wire0, wire1)

        return _Circ
```
Because the way `magma` works, `_Circ` has to be a local class, which cannot be pickled using native Python `pickle` package. Then I switched to `pathos` for the multiprocessing pickling.

It runs fine until it tries to pickle the magma types, which it throws the following exception:
```
    def save_global(self, obj, name=None):
        write = self.write
        memo = self.memo
    
        if name is None:
            name = getattr(obj, '__qualname__', None)
        if name is None:
            name = obj.__name__
    
        module_name = whichmodule(obj, name)
        try:
            __import__(module_name, level=0)
            module = sys.modules[module_name]
            obj2, parent = _getattribute(module, name)
        except (ImportError, KeyError, AttributeError):
            raise PicklingError(
                "Can't pickle %r: it's not found as %s.%s" %
>               (obj, module_name, name)) from None
E           _pickle.PicklingError: Can't pickle Array[1, Bits[1, Bit]]: it's not found as magma.array.Array[1, Bits[1, Bit]]
```

It seems to be that the entire `magma` is not designed for multi-processing environment. In this case we have to be very creative on how to speed up the process.
1. Implement copy on write (COW). This may takes some time to implement but suitable for single-processor environment as in `magma`.
2. Allow designer to by-pass all the uniquification check if they know what they are doing. This sounds like a hack, but should speed up the processing significantly.
3. Rewrite the magma backend in C++, this will at least give us about 50x speedup.

alexcarsello, 2019-04-28:
Update: The run is still not complete. I am confident that most of the slowdown was caused by the introduction of the peak PE core. I'll have to stub it out for now when creating the full size CGRA.

alexcarsello, 2019-04-29:
One more update: time to generate a 32x16 CGRA with the Peak PE core stubbed out is approximately 1 hour, so most of the increase was caused by the introduction of the Peak PE core.. However, 1 hour is still very long, so there are other issues that need to be addressed as well.

leonardt, 2019-04-29:
Compiling the peak PE core has some low hanging fruit for speedup, right now it's dumping out Python files and reading them in to compile them (this was for easy debugging). We should change that to be a "debug" mode and keep the generated Python code in memory by default. We should also be able to refactor the compiler pipeline to avoid serializing/deserializing the AST between stages (sequential -> combinational -> base magma). For the 1 hour case with the Peak PE stubbed out, it would be good to get a better understanding of where time is being spent (garnet, magma, coreir) and in what parts so we know where the optimization effort should be placed. Is there any Python cprofile data we can look at it? (https://docs.python.org/3.7/library/profile.html)

leonardt, 2019-04-29:
Re bypassing uniquification check: is there a way to implement cacheing/memoization at the higher level (e.g. garnet or gemstome)? This is effectively "bypassing" uniquification (by generating a reference to the same circuit, rather than redefining the circuit and relying on uniquification).

leonardt, 2019-04-29:
Re: Rewrite the magma backend in C++, this will at least give us about 50x speedup.

Raj and I discussed a refactor of the magma internals to use coreir as the IR rather than magma's own IR. There's some implications in terms of useability (I think it will be harder to implement good error messages and debugging the internals), but it should offer a performance increase when constructing graphs (dispatch to coreir API calls instead of Python) and skip the backend compilation phase (coreir representation already exists).

However, we also identified some low hanging fruit in the magma IR if we want to preserve the Python layer, for example, not generating the leaf elements of the recursive types unless required (i.e. lazy construction of all the `Bit` objects at the leaf of a tuple of arrays). CoreIR supports describing bulk connections (e.g. tuple -> tuple) without having to ever realize the leaf elements of the bulk type, so we could defer the instantiation and wiring of those elements until the user specifically references them (e.g. selects a bit out of an array) and instead do the wiring on the bulk type level (connect arrays to arrays). This still requires some significant changes to the IR because right now magma assumes all connections are done on the bit level (I think this is just the simplest, most general way to implement this logic, so it makes sense for a first pass, but i don't think it precludes these optimizations for bulk connections)

leonardt, 2019-04-29:
I'm curious at exactly why generating the PE core adds such a significant multiplier to the runtime (1 hour vs over 7 hours). Does this imply that the PE core is being compiled to magma multiple times? (e.g. once per 32x16 tile) Is there any way to avoid rerunning the lassen->RTL compile step more than once (assuming this already isn't the case)? I know that process is slower than it could be, but the runtime for that shouldn't be affected by scaling the number of times it needs to instanced. Is it that the entire 7 hours is compiling the PE core once? (That seems unlikely to me)

Kuree, 2019-04-29:
@leonardt How feasible is to make magma multi-processing friendly? This doesn't solve the root problem but will at least give us some significant speed up. If it's a simple fix in magma I can work on the rest.

leonardt, 2019-04-29:
what exactly is the issue? Is it that a magma circuit needs to support pickling? If so, let me investigate what needs to be done to support that

Kuree, 2019-04-29:
From the error message I got, it appears so. I'm not sure if there is more. I'm happy to investigate to possibility to support multi-processing.

rsetaluri, 2019-05-06:
Keeping track of latest performance issues in #227. Closing.

closed 2019-05-06 by rsetaluri