StanfordAHA/garnet #225: Verilog Generation Failure

alexcarsello, 2019-05-02:
Getting the following exception when I try to generate the verilog for the 32x16 CGRA:

File "garnet.py", line 190, in <module>
   main()
 File "garnet.py", line 179, in main
   magma.compile("garnet", garnet_circ, output="coreir-verilog")
 File "/sim/ajcars/magma/magma/compile.py", line 114, in compile
   uniquification_pass(main, uniquification_mode)
 File "/sim/ajcars/magma/magma/uniquification.py", line 102, in uniquification_pass
   pass_.run()
 File "/sim/ajcars/magma/magma/uniquification.py", line 79, in run
   super(UniquificationPass, self).run()
 File "/sim/ajcars/magma/magma/passes/passes.py", line 58, in run
   self._run(self.main)
 File "/sim/ajcars/magma/magma/uniquification.py", line 70, in _run
   self._run( instancedefinition )
 File "/sim/ajcars/magma/magma/uniquification.py", line 70, in _run
   self._run( instancedefinition )
 File "/sim/ajcars/magma/magma/uniquification.py", line 70, in _run
   self._run( instancedefinition )
 [Previous line repeated 2 more times]
 File "/sim/ajcars/magma/magma/uniquification.py", line 76, in _run
   self(definition)
 File "/sim/ajcars/magma/magma/uniquification.py", line 54, in __call__
   type(definition).rename(definition, new_name)
 File "/sim/ajcars/magma/magma/circuit.py", line 149, in rename
   raise Exception("Can not rename a verilog wrapped file")
Exception: Can not rename a verilog wrapped file

Kuree, 2019-05-02:
This is using the newest `magma` master branch. How to reproduce:

````
git clone https://github.com/StanfordAHA/garnet && cd garnet
pip install -r requirements.txt
# use the master branch
pip install -e git+git://github.com/phanrahan/magma#egg=magma
python garnet.py -v
```

rsetaluri, 2019-05-02:
Ok, I will work on fixing this. To unblock yourself, you can check out `c3030d574ff32633407de5d2402969758ec4d7e6` in magma?

Also, can you tell me which definition it is trying to rename?

alexcarsello, 2019-05-02:
This error message is all I have... It's not clear what it's trying to rename.

rsetaluri, 2019-05-02:
Does checking out the old commit unblock you?

alexcarsello, 2019-05-03:
yes that unblocks for now. Though the uniquification fix is not there. It would be very helpful to merge that branch in as well.

Kuree, 2019-05-03:
More errors: https://github.com/phanrahan/magma/issues/389

rsetaluri, 2019-05-06:
Is this still an issue?

closed 2019-05-13 by rsetaluri

Kuree, 2019-05-13:
Not any more.