StanfordAHA/garnet #252: New memory core cannot be compiled on ncsim

Kuree, 2019-05-24:
```
assign valid_gate = (stencil_width == 0) ? 1 : vg_ctr >= (stencil_width - 1);
--
  | \|
  | ncvlog: *E,RANOTL (linebuffer_control_unq1.v,90\|16): A reg is not a legal lvalue in this context [6.1.2(IEEE)].
  | module worklib.linebuffer_control_unq1:v
  | errors: 1, warnings: 0
  | file: fifo_control_unq1.v
  | reg [15:0] data_in [1:0];

```

Can you change your logic into system verilog and use `logic` instead?

Kuree, 2019-05-24:
Complete error log. 78 errors in total:

```

file: linebuffer_control_unq1.v
--
  | assign valid_gate = (stencil_width == 0) ? 1 : vg_ctr >= (stencil_width - 1);
  | \|
  | ncvlog: *E,RANOTL (linebuffer_control_unq1.v,90\|16): A reg is not a legal lvalue in this context [6.1.2(IEEE)].
  | module worklib.linebuffer_control_unq1:v
  | errors: 1, warnings: 0
  | file: fifo_control_unq1.v
  | reg [15:0] data_in [1:0];
  | \|
  | ncvlog: *E,DUPIDN (fifo_control_unq1.v,130\|17): identifier 'data_in' previously declared [12.5(IEEE)].
  | module worklib.fifo_control_unq1:v
  | errors: 1, warnings: 0
  | file: sram_control_unq1.v
  | input [15:0] mem_to_sram_data [1:0];
  | \|
  | ncvlog: *E,EXPSMC (sram_control_unq1.v,83\|30): expecting a semicolon (';') [12.3.2(IEEE)].
  | output [15:0] sram_to_mem_data [1:0];
  | \|
  | ncvlog: *E,EXPSMC (sram_control_unq1.v,86\|31): expecting a semicolon (';') [12.3.2(IEEE)].
  | output [8:0] sram_to_mem_addr [1:0];
  | \|
  | ncvlog: *E,EXPSMC (sram_control_unq1.v,89\|30): expecting a semicolon (';') [12.3.2(IEEE)].
  | module worklib.sram_control_unq1:v
  | errors: 3, warnings: 0
  | file: doublebuffer_control_unq1.v
  | output reg [15:0]  doublebuffer_data_in  [1:0];
  | \|
  | ncvlog: *E,MEMDIO (doublebuffer_control_unq1.v,124\|38): Memory 'doublebuffer_data_in' previously declared as input/output/inout.
  | output reg [8:0]  doublebuffer_addr_mem [1:0];
  | \|
  | ncvlog: *E,MEMDIO (doublebuffer_control_unq1.v,127\|38): Memory 'doublebuffer_addr_mem' previously declared as input/output/inout.
  | module worklib.doublebuffer_control_unq1:v
  | errors: 2, warnings: 0
  | file: sram_stub_unq1.v
  | module worklib.sram_stub_unq1:v
  | errors: 0, warnings: 0
  | file: mem_unq1.v
  | module worklib.mem_unq1:v
  | errors: 0, warnings: 0
  | file: memory_core.v
  | mem_data_out[0]};
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,252\|13): 'mem_data_out': undeclared identifier [12.5(IEEE)].
  | mem_data_out[0]};
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,254\|13): 'mem_data_out': undeclared identifier [12.5(IEEE)].
  | mem_data_out[1]};
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,256\|13): 'mem_data_out': undeclared identifier [12.5(IEEE)].
  | mem_data_out[1]};
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,258\|13): 'mem_data_out': undeclared identifier [12.5(IEEE)].
  | mem_cen_int[0] = 1'b1; //'
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,262\|16): 'mem_cen_int': undeclared identifier [12.5(IEEE)].
  | mem_cen_int[1] = 1'b1; //'
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,263\|16): 'mem_cen_int': undeclared identifier [12.5(IEEE)].
  | mem_wen[0] =  (config_write & (
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,264\|12): 'mem_wen': undeclared identifier [12.5(IEEE)].
  | mem_wen[1] =  (config_write & (
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,267\|12): 'mem_wen': undeclared identifier [12.5(IEEE)].
  | mem_ren[0] =  (config_read & (
[... 188 lines cut ...]
  | assign mem_cen[0] = mem_cen_int[0] & ( mem_wen[0] \| mem_ren[0]);
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,393\|58): 'mem_ren': undeclared identifier [12.5(IEEE)].
  | assign mem_cen[1] = mem_cen_int[1] & ( mem_wen[1] \| mem_ren[1]);
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,394\|45): 'mem_wen': undeclared identifier [12.5(IEEE)].
  | assign mem_cen[1] = mem_cen_int[1] & ( mem_wen[1] \| mem_ren[1]);
  | \|
  | ncvlog: *E,UNDIDN (memory_core.v,394\|58): 'mem_ren': undeclared identifier [12.5(IEEE)].
  | .num_words_mem(num_words_mem_fifo_to_lb),
  | \|
  | ncvlog: *E,DUPIMP (memory_core.v,418\|38): Identifier 'num_words_mem_fifo_to_lb', implicitly declared here, first as a wire, is subsequently redeclared.
  | wire [12:0] num_words_mem_fifo_to_lb;
  | \|
  | ncvlog: *E,DUPIDN (memory_core.v,437\|35): identifier 'num_words_mem_fifo_to_lb' previously declared [12.5(IEEE)].
  | .mem_to_fifo_data(mem_data_out),
  | \|
  | ncvlog: *E,DUPIMP (memory_core.v,458\|29): Identifier 'mem_data_out', implicitly declared here, first as a wire, is subsequently redeclared.
  | wire [15:0] mem_data_out [1:0];
  | \|
  | ncvlog: *E,DUPIDN (memory_core.v,544\|23): identifier 'mem_data_out' previously declared [12.5(IEEE)].
  | module worklib.memory_core:v
  | errors: 78, warnings: 0

```

mbstrange2, 2019-05-24:
Submitted commits to men_core_fixes

closed 2019-05-25 by Kuree