StanfordAHA/garnet #263: Genus segfaults during power aware synthesis of memory tile

alexcarsello, 2019-05-27:
Genus is throwing a segfault when I try to synthesize the new memory tile with the PWR_AWARE flag set to 1. It works when I set the flag to 0.

ankita0805, 2019-05-27:
I will take a look. Is it reproducible using Garnet master branch for the memory tile RTL generation and tapeout branch for the scripts?

alexcarsello, 2019-05-27:
Yes. I've actually merged the latest master into tapeout, so you can just use tapeout to reproduce it. Just remember to instantiate the TSMC memory in mem_unq1.v.

ankita0805, 2019-05-27:
https://github.com/StanfordAHA/garnet/pull/264

Please update one constraint that's missing in the tapeout branch to match the updated Mux RTL . 

Synthesis has crossed the point where it was crashing. The run will take time -  I will let it run till the end and confirm it completes fine.

alexcarsello, 2019-05-27:
Awesome thanks for the quick fix! Feel free to close the issue and merge the PR once the run finishes. Also, if you are running with the initial timing constraints (2 ns), you may want to slow it down to 5-6 ns

ankita0805, 2019-05-27:
I doubled the initial constraint, and it's slow. I am trying 8 ns now just to make sure it completes as expected.

alexcarsello, 2019-05-27:
I was running it at 6 ns successfully with PWR_AWARE set to 0. I've discussed this with Max... we need to make it faster before tapeout.

closed 2019-05-27 by alexcarsello