StanfordAHA/garnet #286: Unused 1bit data coming out from cgra io_tile should set to 0 after configuration.

kong0329, 2019-05-30:
1bit data coming out from `cgra_io_tile` are connected to `io_controller` in `global buffer` and used as control signals such as `write_en` or `read_en`.
For example, in 4x4 cgra case, even if application only use `io_tile_x00_y00` and `io_tile_x01_y00` as input or output, I want all 1bit signals coming from `io_tile_x02_y00` and `io_tile_x03_y00` to be 0.
Can we have mux with default constant value  so that all unused ports drive `0` after configuration?

kong0329, 2019-05-30:
Close for now.

closed 2019-05-30 by kong0329

Kuree, 2019-05-31:
Why it should be the case? That means I have to generate lots of extra
routing and use lots of extra PE tiles.

kong0329, 2019-05-31:
No. If we have mux with constant value like 0, we only need to increase the number of inputs to connection box.
We do not need extra routing or PE tiles.
This is also related to the issue that we had before, where `flush` signal to memory tile just randomly goes high. We did not config so `flush` signal stays `0` because we were not using `flush`, which ends up having `flush` go high.

For controllers in global buffer, I will just have some control logics to ignore values if it is unused, so i close it for now.