StanfordAHA/garnet #335: Run pd flow for PE tile on each check-in

steveri, 2019-08-06:
The assignment---from Priyanka email 2 aug 2019
> Alex, Steve is willing to help run the scripts in virtual tapeout
> mode. We should start running portions of the flow in this mode. For
> example, to start with, we can run PE and memory tile flows when pull
> requests on those modules happen. Could you please work with Steve to
> set that up? We can run the existing scripts now, and gradually
> refine the scripts to make them more modular.

steveri, 2019-08-06:
Notes - This README file contains instructions for building the hard macro:
https://github.com/StanfordAHA/garnet/blob/tapeout/tapeout_16/README.md

Sub-challenges:
1. Figure out how buildkite works
2. Fix garnet because currently, in travis, *ALL BRANCHES ARE BROKEN* :(
https://travis-ci.com/StanfordAHA/garnet

alexcarsello, 2019-08-06:
This readme is actually out of date.... I’ll update it this morning.

alexcarsello, 2019-08-06:
Both synthesis and P&R instructions should be up to date now. Let me know if you have any issues.

steveri, 2019-08-07:
**STATUS** I've started on a script  `garnet/tapeout_16/do_pe.sh`  as a prototype for the script(s) that will run on buildkite at each check-in. The script will live in my branch   tapeout_sr   until I get it working at least minimally. The basic idea is to run through/verify the garnet setup procedure https://github.com/StanfordAHA/garnet/blob/master/README.md

and then do synthesis and PNR as outlined in https://github.com/StanfordAHA/garnet/blob/tapeout/tapeout_16/README.md

Some of the problems I've run into so far:
- missing python packages (easy fix)
- missing pip to install python packages (more annoying fix)
- wrong default python2 vs. python3 (another annoying fix)
- wrong python3 version---peak wants 3.7 and we have 3.6 (not fixed yet, maybe that's next)

I'm surprised the garnet.py has apparently never yet run on the arm7 machine(?) I'm guessing that up til now, we ran garnet on the TSMC machine(?) and then copied the result to the arm7 machine?

**Notes on garnet version control**
What are the rules for updating garnet's "tapeout" branch? It looks like we are continuing to update it, so I assume that when/if we need a snapshot from the actual original tapeout we will just find it by checking out the appropriate commit. Meanwhile, I guess I will try and be a good citizen and stage my changes through a sub-branch "tapeout_sr"

Also: garnet currently has 50 branches, at least four of which seem to be "tapeout" branches...I filed an issue https://github.com/StanfordAHA/garnet/issues/337

Also: is there a reason why the tapeout branch wouldn't merge with master at some point?

steveri, 2019-08-23:
I have successfully run PE and Mem flows in buildkite see e.g. this mem run
https://buildkite.com/tapeout-aha/test/builds/63
...and this PE run
https://buildkite.com/tapeout-aha/test/builds/61

Both have inconsequential errors related to missing (but unused!) sram library cells. I think the next step will be to fire up separate pipelines, one each for PE and Mem, on each check-in, at which point I will close this issue.

steveri, 2019-09-04:
Closing this issue b/c
1. One-button PE tile flow works pretty well now
 2. We've moved on to top level flow, a superset goal which includes PE and mem tiles as well
For furhter updates see task list https://github.com/StanfordAHA/garnet/issues/340

closed 2019-09-04 by steveri