StanfordAHA/garnet #336: MASTER BRANCH FAILING IN TRAVIS (libverilogAST missing)

steveri, 2019-08-06:
Master branch has been failing for almost a month now. Why is no one fixing this? Current problem seems to be here:

> E   OSError: libverilogAST.so: cannot open shared object file: No such file or directory
> The command "docker exec -i garnet bash -c "/garnet/.travis/run.sh"" exited with 4.

https://travis-ci.com/StanfordAHA/garnet/builds/122010364

Kuree, 2019-08-06:
I will pin all the repos that don't have backward compatibilities.

steveri, 2019-08-06:
@Kuree - Do we actually need libverilogAST? I assumed it was something you added, which is why you were assigned initially...but now that I look at it I'm thinking it might be a CoreIR thing...should we reassign to @rdaly525 ...?

steveri, 2019-08-06:
To be clear: this is the error that I see in the travis log:

> E   OSError: libverilogAST.so: cannot open shared object file: No such file or directory
> The command "docker exec -i garnet bash -c "/garnet/.travis/run.sh"" exited with 4.

Kuree, 2019-08-06:
This is due to the new changes in coreir and incorrect build setup in pycoreir. pycoreir builds did not produce a valid wheel (in fact, no wheel is produced for linux: https://pypi.org/project/coreir/2.0.21/#files).

I will hammer down all the breaking changes to return garnet into green state. We can discuss how to maintain garnet tomorrow.

Kuree, 2019-08-06:
Should be fixed by https://github.com/StanfordAHA/garnet/commit/9cd1ab7c01ade95fae8db6ad625cb9d548934fbf.

closed 2019-08-06 by Kuree