StanfordAHA/garnet #337: garnet has fifty branches (ish)

steveri, 2019-08-07:
Branches are outta control

```
% git branch -va
HEAD                  -> origin/master
add-aoi-mux           05d3e0d Fix magma const wiring
app_test              f184725 Try to run simple commands
auto_restart          839f909 Update auto restart
axi                   4cdca00 Update README.md
cache_genesis         92e4727 fix other genesis tests
cache_mem             b7a75cd cache memory core creation
chain_test            4d658a6 Merge remote-tracking branch 'origin/master' into chain_test
db_shrink             1856b40 print out config reg plain for Alex
db_unified            b37a4ae removed LB mode from linebuffer test, ...
flow                  c1c677c Merge remote-tracking branch 'origin/mem_cen_patch' into flow
genesis-improvements  480d645 Fix gc generator
glb_2d                de16ea0 Remove dump
glb_impl              f3ccab9 Merge remote-tracking branch 'origin/master' into glb_impl
global_buffer         6f58230 Finish global buffer magma
global_controller_axi 86987d3 Add axi controller to global controller
halide-flow           c0033db merge master into halide-flow
hotfix_remote         cb28bd2 fix the remote changes
interconnect-verify   825b0b8 local tests passed. need to debug on travis
late-bind-peak-rtl    19881c5 fix typo
master                9cd1ab7 skip broken tests
mem_cen_patch         44724b5 Update doublebuffer_control.svp
mem_core_wire         4866e3a Try removing memory core wire statements
memory-core-272       e631389 Improved FIFO test
memory_core_impl      a71a9bc garnet
new_genesis           78f7051 use master branch of magma
pd-feature            4912013 fix pep8 in pd_config
pd-passes             67e21a1 pycodestyle check fix
pd-updates            078c944 change regexp
pd_mux_transform      7219c94 select pin fix
pd_pass_or            1104027 point gemstone remote back to master
pin_versions          9cd1ab7 skip broken tests
powerdomains          ff1b516 test pass logic
powerdomains-2        0123238 pycodestyle check fix
pulse                 ddc6cc2 fix transition to inactive when handling auto restart
refactor_mapper       aaf774c start to refactor the mapper
reg_passes            a44daf6 point to master canal
remove_genesis        395a2f1 fix travis
simple_mapper         6a6a3bc Merge branch 'master' into simple_mapper
simple_mapper_chain   f1bd655 finish up chain logic
simple_mapper_inputs  23c35c2 add multiple en ports
tapeout               c60283b Update README.md
tapeout-pd            39278cb PD related changes
tapeout-test          c7870b6 Update dummy metal insertion script
tapeout_kty           81c469a Add makefile
tapeout_lvs           0e734ab Fix DRC on tile level
test_master           6bfe5ac print out the info
ub_merge              9a6b8fc hotfix: adjust memory pnr priority
update-bv-syntax      d692fa1 Update BitVector constructors
vdd-sw-updates        ff71470 remove M7-M9 VDD_SW
```

Kuree, 2019-08-07:
Git remote only tracks 26 branches at the time of writing. I think you're referring some local branches as well. I will start to clean out some stale branches.

steveri, 2019-08-07:
Oops, you are right only 25-ish branches. Apparently I have to do this, every once in awhile:
`git fetch origin --prune`

Still too many. If you (or anyone else) has stale branches they can clean up, please do so, the community will thank you I'm sure :)

mbstrange2, 2019-08-09:
cleared all of mine