StanfordAHA/garnet #341: Why does mapping use so much memory?

THofstee, 2019-08-16:
I can't map applications locally because I run out of memory. It feels to me like mapping things with garnet is creating a fully realized instantiation of the design which takes a massive amount of memory. I don't see why this is necessary.

If you need to get information out of the modules in the design for mapping, you only need thin shells of the instances. A PE should know the addresses of its configuration registers and how to configure them.

Instead, I'm seeing mapping take upwards of 2500MB of memory before the mapping itself even starts.

THofstee, 2019-08-16:
Also, at the very least a batch mode would be useful for testing so I only need to spend the time waiting for garnet to set itself up once for all my apps, instead of waiting every time I map something.

THofstee, 2019-08-16:
For reference, this is a garnet issue, not a mapper or pnr issue.

```
garnet: 1:06.15 elapsed 103%CPU 1083956kB max resident
mapper: 0:00.21 elapsed  99%CPU   36032kB max resident
placer: 0:00.08 elapsed 106%CPU   10800kB max resident
router: 0:01.17 elapsed  97%CPU   64712kB max resident
```

steveri, 2019-08-21:
Assigning @Kuree and @rdaly525 at least long enough to ask this question: are we currently using Ross' auto-mapper or Keyi's custom mapper?

Kuree, 2019-08-21:
Raj has made a decision during the early design stage that everything should be done in memory while garnet is running. As a result, we have to construct everything in order to map the applications.

Here are two ways I can think of to deal with that:
1. Hook up http module to have garnet running as a web server when applications are being mapped. This won't solve the memory issue, but will solve the performance issue as we only need to construct the circuit once.
2. Implement a hacky standalone process to map. My PnR and mapper follows the unix design philosophy, so it the only step needed is to rip the interconnect generator (the digraph generator) out of garnet and only use that to do bitstream configuration. This still takes some time since generating peak is not that fast.

steveri, 2019-08-21:
Removing @rdaly525 from assignments b/c it's the custom mapper...