StanfordAHA/garnet #342: "official" layout_Tile script does not work with innovus 19

steveri, 2019-08-21:
This is probably an easy problem to solve but I haven't yet tackled it.

Currently we do layout using an old script from Alex's directory
`/sim/ajcars/garnet/tapeout_16/scripts/layout_Tile.tcl`
instead of the latest/greatest found in github master. I.e. my version of  "run_layout.csh" (in branch tapeout_sr) does this
```
    # set s=../../scripts/layout_Tile.tcl # oops does not work w/innovus 19
    set s=/sim/ajcars/garnet/tapeout_16/scripts/layout_Tile.tcl
    innovus -no_gui -abort_on_error -replay $s

```
The problem seems to be that we switched to a newer version of innovus at the last minute and the newer tcl script does not work with the newer innovus.

I think this will be easy to debug but have not gotten around to it yet. Here is the error:
```
**ERROR: (IMPSE-110): File '../../scripts/layout_Tile.tcl' line 157:
  ambiguous command name "placeInst": placeInstance
  placeInstsAtFenceCenter.

```
Which do we intend, placeInstance or placeInstanceAtFenceCenter?
I'm guessing placeInstance, and when/if I fix it that is what I [steveri] would try first.

Your comments/help more than welcome!

THofstee, 2019-08-22:
@alexcarsello told me to change them to `placeInstance` when I was trying to run the scripts earlier last week. It worked fine, so I imagine that's the one you want.

steveri, 2019-08-22:
Thanks Teguh! Yes, I changed them to PlaceInstance and it seems to be working fine now...all my new scripts are using this "official" version now, along with innovus v19

closed 2019-08-22 by steveri