StanfordAHA/garnet #345: run_layout errors: "cannot find [sram] library cells"

steveri, 2019-08-23:
synthesis and layout runs without error, mostly, except for several errors related to missing sram libraries:
https://buildkite.com/tapeout-aha/test/builds/63
`[08/22 14:41:14  27471s] *** Message Summary: 16612 warning(s), 8 error(s)`
```
...
**ERROR: (TCLCMD-917): Cannot find 'library cells' that match
  'ts1n16ffcllsblvtc256x32m4sw_ssgnp0p72vm40c/TS1N16FFCLLSBLVTC256X32M4SW'
  (File results_syn/syn_out._default_constraint_mode_.sdc, Line
  15907).
...
```

The source of the error seems to be this list of `"set_dont_use"` commands in the indicated script `"syn_out._default_constraint_mode_.sdc"`

```
tail synth/Tile_Mem*/results_syn/syn_out._default_constraint_mode_.sdc
  set_dont_use [get_lib_cells
    ts1n16ffcllsblvtc256x32m4sw_ssgnp0p72vm40c/
    TS1N16FFCLLSBLVTC256X32M4SW]
  set_dont_use [get_lib_cells
    ts1n16ffcllsblvtc256x32m8sw_ssgnp0p72vm40c/
    TS1N16FFCLLSBLVTC256X32M8SW]
  ...
```
There appear to be four missing cells; for some reason, each missing library cell generates two errors, thus eight errors total related to this problem.

steveri, 2019-08-23:
`synthesis.tcl` script defines 5 sram library cells, four of which go on to cause the indicated errors. 

```
set_attr library [list \
/sim/ajcars/mc/ts1n16ffcllsblvtc512x16m8s_130a\NLDM/ts1n16ffcllsblvtc512x16m8s_130a_ssgnp0p72vm40c.lib \
/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m4sw_130a\NLDM/ts1n16ffcllsblvtc256x32m4sw_130a_ssgnp0p72vm40c.lib \
/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m8sw_130a\NLDM/ts1n16ffcllsblvtc256x32m8sw_130a_ssgnp0p72vm40c.lib \
/sim/ajcars/mc/ts1n16ffcllsblvtc2048x32m8sw_130a\NLDM/ts1n16ffcllsblvtc2048x32m8sw_130a_ssgnp0p72vm40c.lib \
/sim/ajcars/mc/ts1n16ffcllsblvtc2048x64m8sw_130a\NLDM/ts1n16ffcllsblvtc2048x64m8sw_130a_ssgnp0p72vm40c.lib \
] 
```

I guess the script is generating `"dont_use"` commands for each of the four unused sram cells

```
tail results_syn/syn_out._default_constraint_mode_.sdc
 15907  set_dont_use [get_lib_cells ts1n16ffcllsblvtc256x32m4sw...
 15908  set_dont_use [get_lib_cells ts1n16ffcllsblvtc256x32m8sw...
 15909  set_dont_use [get_lib_cells ts1n16ffcllsblvtc2048x32m8sw...
 15910  set_dont_use [get_lib_cells ts1n16ffcllsblvtc2048x64m8sw...

```

Meanwhile, in the `layout_Tile.tcl` script, only the 512x16 sram is included in the list of lef_files

```
set lef_file [list /tsmc16/download/TECH16FFC/N16FF_PRTF_Cad_1.2a/PR_tech/Cadence/LefHeader/Standard/VHV/N16_Encounter_9M_2Xa1Xd3Xe2Z_UTRDL_9T_PODE_1.2a.tlef \
  /sim/ajcars/mc/ts1n16ffcllsblvtc512x16m8s_130a/LEF/ts1n16ffcllsblvtc512x16m8s_130a_m4xdh.lef \
  /tsmc16/TSMCHOME/digital/Back_End/lef/tcbn16ffcllbwp16p90pm_100a/lef/tcbn16ffcllbwp16p90pm.lef \
  /tsmc16/TSMCHOME/digital/Back_End/lef/tcbn16ffcllbwp16p90_100a/lef/tcbn16ffcllbwp16p90.lef ]

```

My proposed fix: add the four missing sram library cells to the `layout_Tile.tcl` script:

```
diff --git a/tapeout_16/scripts/layout_Tile.tcl b/tapeout_16/scripts/layout_Tile.tcl
index a7c848a..f7f456c 100755
--- a/tapeout_16/scripts/layout_Tile.tcl
+++ b/tapeout_16/scripts/layout_Tile.tcl
@@ -1,5 +1,9 @@
 set lef_file [list /tsmc16/download/TECH16FFC/N16FF_PRTF_Cad_1.2a/PR_tech/Cadence/LefHeader/Standard/VHV/N16_Encounter_9M_2Xa1Xd3Xe2Z_UTRDL_9T_PODE_1.2a.tlef \
 /sim/ajcars/mc/ts1n16ffcllsblvtc512x16m8s_130a/LEF/ts1n16ffcllsblvtc512x16m8s_130a_m4xdh.lef \
+/sim/ajcars/mc/ts1n16ffcllsblvtc2048x32m8sw_130a/LEF/ts1n16ffcllsblvtc2048x32m8sw_130a_m4xdh.lef \
+/sim/ajcars/mc/ts1n16ffcllsblvtc2048x64m8sw_130a/LEF/ts1n16ffcllsblvtc2048x64m8sw_130a_m4xdh.lef \
+/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m4sw_130a/LEF/ts1n16ffcllsblvtc256x32m4sw_130a_m4xdh.lef \
+/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m8sw_130a/LEF/ts1n16ffcllsblvtc256x32m8sw_130a_m4xdh.lef \
 /tsmc16/TSMCHOME/digital/Back_End/lef/tcbn16ffcllbwp16p90pm_100a/lef/tcbn16ffcllbwp16p90pm.lef \
 /tsmc16/TSMCHOME/digital/Back_End/lef/tcbn16ffcllbwp16p90_100a/lef/tcbn16ffcllbwp16p90.lef ]
```

steveri, 2019-08-23:
Fix described above reduced the number of errors from eight down to four. Layout script still complains about the two 256x32 sram cells:

```
**ERROR: (TCLCMD-917): Cannot find 'library cells' that match
  'ts1n16ffcllsblvtc256x32m4sw_ssgnp0p72vm40c/TS1N16FFCLLSBLVTC256X32M4SW'
  (File results_syn/syn_out._default_constraint_mode_.sdc, Line
  15907).

**ERROR: (TCLCMD-917): Cannot find 'library cells' that match
  'ts1n16ffcllsblvtc256x32m8sw_ssgnp0p72vm40c/TS1N16FFCLLSBLVTC256X32M8SW'
  (File results_syn/syn_out._default_constraint_mode_.sdc, Line 15908)
```

steveri, 2019-08-23:
Looks like scripts `mmmc.tcl` and `mmode.tcl` both are missing the 256x32 cells from their library list(s) (see below). My proposed fix: add the missing cells to mmmc.tcl and see what happens.

```
diff --git a/tapeout_16/scripts/mmmc.tcl b/tapeout_16/scripts/mmmc.tcl
index 42b8119..90f9c16 100644
--- a/tapeout_16/scripts/mmmc.tcl
+++ b/tapeout_16/scripts/mmmc.tcl
@@ -5,6 +5,8 @@ create_library_set -name ss_0p72_m40c_lib_set -timing [list \
 /sim/ajcars/mc/ts1n16ffcllsblvtc512x16m8s_130a/NLDM/ts1n16ffcllsblvtc512x16m8s_130a_ssgnp0p72vm40c.lib \
 /sim/ajcars/mc/ts1n16ffcllsblvtc2048x32m8sw_130a/NLDM/ts1n16ffcllsblvtc2048x32m8sw_130a_ssgnp0p72vm40c.lib \
 /sim/ajcars/mc/ts1n16ffcllsblvtc2048x64m8sw_130a/NLDM/ts1n16ffcllsblvtc2048x64m8sw_130a_ssgnp0p72vm40c.lib \
+/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m4sw_130a/NLDM/ts1n16ffcllsblvtc256x32m4sw_130a_ssgnp0p72vm40c.lib \
+/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m8sw_130a/NLDM/ts1n16ffcllsblvtc256x32m8sw_130a_ssgnp0p72vm40c.lib \
 /tsmc16/TSMCHOME/digital/Front_End/timing_power_noise/NLDM/tcbn16ffcllbwp16p90pm_100a/tcbn16ffcllbwp16p90pmssgnp0p72vm40c.lib \
 ]
 
@@ -13,6 +15,8 @@ create_library_set -name ss_0p72_125c_lib_set -timing [list \
 /sim/ajcars/mc/ts1n16ffcllsblvtc512x16m8s_130a/NLDM/ts1n16ffcllsblvtc512x16m8s_130a_ssgnp0p72v125c.lib \
 /sim/ajcars/mc/ts1n16ffcllsblvtc2048x32m8sw_130a/NLDM/ts1n16ffcllsblvtc2048x32m8sw_130a_ssgnp0p72v125c.lib \
 /sim/ajcars/mc/ts1n16ffcllsblvtc2048x64m8sw_130a/NLDM/ts1n16ffcllsblvtc2048x64m8sw_130a_ssgnp0p72v125c.lib \
+/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m4sw_130a/NLDM/ts1n16ffcllsblvtc256x32m4sw_130a_ssgnp0p72v125c.lib \
+/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m8sw_130a/NLDM/ts1n16ffcllsblvtc256x32m8sw_130a_ssgnp0p72v125c.lib \
 /tsmc16/TSMCHOME/digital/Front_End/timing_power_noise/NLDM/tcbn16ffcllbwp16p90pm_100a/tcbn16ffcllbwp16p90pmssgnp0p72v125c.lib \
 ]
 
@@ -21,6 +25,8 @@ create_library_set -name ff_0p88_0c_lib_set -timing [list \
 /sim/ajcars/mc/ts1n16ffcllsblvtc512x16m8s_130a/NLDM/ts1n16ffcllsblvtc512x16m8s_130a_ffgnp0p88v0c.lib \
 /sim/ajcars/mc/ts1n16ffcllsblvtc2048x32m8sw_130a/NLDM/ts1n16ffcllsblvtc2048x32m8sw_130a_ffgnp0p88v0c.lib \
 /sim/ajcars/mc/ts1n16ffcllsblvtc2048x64m8sw_130a/NLDM/ts1n16ffcllsblvtc2048x64m8sw_130a_ffgnp0p88v0c.lib \
+/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m4sw_130a/NLDM/ts1n16ffcllsblvtc256x32m4sw_130a_ffgnp0p88v0c.lib \
+/sim/ajcars/mc/ts1n16ffcllsblvtc256x32m8sw_130a/NLDM/ts1n16ffcllsblvtc256x32m8sw_130a_ffgnp0p88v0c.lib \
 /tsmc16/TSMCHOME/digital/Front_End/timing_power_noise/NLDM/tcbn16ffcllbwp16p90pm_100a/tcbn16ffcllbwp16p90pmffgnp0p88v0c.lib \
 ]
```

priyanka-raina, 2019-08-23:
I suggest going the other way. Remove the unneeded srams from the library. Ultimately, we want to have gemstone generate individual synthesis scripts for pe, mem, gb, cgra, etc with only the sram libraries that each submodule needs. The current method is to have one big script that adds a superset of all libraries needed in each submodule.

steveri, 2019-08-23:
The above fix seems to have worked, all the "missing cell" errors are gone, at least in my local test. Next I will roll out the fixes to my development branch and see how we do in buildkite...

steveri, 2019-08-23:
@priyanka-raina I agree with everything you just said. I took the quickest/easiest path to short-term success, but of course the correct solution is to never include the unused libraries in the first place. I think we should file this as a separate issue, since it seems like a larger discussion involving how to have a "single source of truth" for memory cells that generates all three (or more?) scripts (`synthesize.tcl, layout_Tile.tcl, mmmc.tcl`...others?) so that they don't end up mismatched...

alexcarsello, 2019-08-23:
It's not a huge deal, but until that feature is actually implemented in Gemstone, I'd actually vote for leaving the SRAMs libraries in. That way we can minimize the number of changes we have to make to PD scripts if we decide to use different size memories, and so in the meantime, while our PD scripts are not being generated, we can use the same scripts for generating blocks with potentially different size memories. 

Also, in terms of maintaining consistency between P&R and synthesis scripts. We should probably refactor the scripts to read all technology libraries from a single file. This will make the scripts easier to maintain, make technology migration easier, and make it easier to add new libraries when needed.

steveri, 2019-08-23:
@alexcarsello I like your solution very much (reading all technology libraries from a single file). I am about to close this issue out when/if the buildkite script runs successfully, but I have opened a new issue
https://github.com/StanfordAHA/garnet/issues/346
specifically about the inconsistency thing, and I am including your suggestion as a/the proposed fix.

steveri, 2019-09-20:
Oops should have closed this out long ago, the errors are indeed long GONE

closed 2019-09-20 by steveri