StanfordAHA/garnet #352: OSError: libverilogAST.so: cannot open shared object file: No such file or directory

fushufeng, 2019-09-13:
When I try to run the commend "pytest", I came up with this error "OSError: libverilogAST.so: cannot open shared object file: No such file or directory", can anyone tell me how to solve this problem ?

THofstee, 2019-09-13:
`pip install pycoreir` should fix it.

leonardt, 2019-09-13:
This is a C++ dependency for coreir, you may need to try installing it directly (either using a release or installing from source: https://github.com/rdaly525/coreir/blob/master/INSTALL.md)

The latest version of pycoreir won't build from source anymore because there was problems having that interoperate with an existing coreir installation.

leonardt, 2019-09-13:
Another option is to install the shared library from source, which should be fairly simple: https://github.com/leonardt/verilogAST-cpp