StanfordAHA/garnet #353: Tapeout: Route script takes forever

steveri, 2019-09-13:
TL;DR Should we skip optDesign step (at least for now)?

It looks like we fixed the slow fill problem https://github.com/StanfordAHA/garnet/issues/351, by doing fill before routing. But now haha it stalls forever in routing. The new script was working on the command `optDesign -postRoute -hold -setup` for over 30 hours before I killed it.

Here's the funny part. It looks like we did the same thing for the actual tapeout back in June. Is that possible? Looking at the logs innovus logs in the final-tapeout directory
`/sim/ajcars/aha-arm-soc-june-2019/components/cgra/garnet/tapeout_16synth_no_pass_through/GarnetSOC_pad_frame_multi_vt`
...it looks like we wrote a snapshot database "final_final" (log24), and then, later, after waiting a very long time for a similar command
`eval_legacy {optDesign -postRoute -setup}` to (not) finish, then, near midnight on June 30, did a ctl-C to halt it. Then it looks like we tried a couple of things that didn't work (log25 and log26), before creating and writing the final netlist and saving everything to final_updated_netlist.db (log27).

Is this correct? Did we perhaps never get optDesign to run on the final design, and/or should I comment out that command in my script in order to get it to finish?

FYI this is what I saw in the logs, in dir
`/sim/ajcars/aha-arm-soc-june-2019/components/cgra/garnet/tapeout_16/synth_no_pass_through/GarnetSOC_pad_frame_multi_vt`

```
------------------------------------------------------------------------
% grep @inn innovus.log24 | less
@innovus 3>    read_db init_route.enc.dat/
@innovus 1400> write_db final_final.db -def -sdc -verilog
@innovus 3346> eval_legacy {optDesign -postRoute -setup}
--- END w/o further writes (killed after 5 hours, see below) ---
[06/30 17:37  36795s] @innovus 3346> eval_legacy {optDesign -postRoute -setup}
[06/30 18:47] #start 2nd optimization iteration ...
[06/30 20:23] #start 4th optimization iteration ...
[06/30 21:46] #start 5th optimization iteration ...
[06/30 22:26] **INFO (INTERRUPT): Current cmd will stop at next legal exit point.
[06/30 22:26] **INFO (INTERRUPT): One more Ctrl-C to exit Innovus ...
[06/30 22:26] Innovus terminated by user interrupt.

------------------------------------------------------------------------
% grep @inn innovus.log25 | less
@innovus 1> read_db pre_eco_deletions.db/
@innovus 23> Initializing DRC checker...
@innovus 27> delete_routes -help
@innovus 28> exit
--- END w/o writing db ---

------------------------------------------------------------------------
% grep @inn innovus.log26 | less
...
@innovus 23> create_net -name RTE_DIG
@innovus 24>
@innovus 24> foreach x [get_property [get_cells {*IOPAD*ext_clk_async* *IOPAD_bottom* *IOPAD_left* *IOPAD_right*}] full_name] {
--- END w/o writing db (killed?) ---

------------------------------------------------------------------------
% grep @inn innovus.log27
@innovus 1> read_db final_final.db/
@innovus 2> create_net -name RTE_DIG
@innovus 3>
@innovus 3> foreach x [get_property [get_cells {*IOPAD*ext_clk_async* *IOPAD_bottom* *IOPAD_left* *IOPAD_right*}] full_name] {
@innovus 4> eval_legacy {source ../../scripts/save_netlist.tcl }
@innovus 5> write_db final_final.db/
@innovus 6> write_db final_updated_netlist.db
@innovus 7> exit
```

steveri, 2019-09-16:
Update: e-mail thread consensus is that of course this step should NOT be skipped, we should instead try and reduce congestion so routing can finish in finite time

```
On 9/13/2019 4:08 PM, Mark wrote:
  The problem was routing congestion in an earlier design step.  We need
  to change the placement of the ICOVL cells, or change the global
  buffer layout to make sure this works.

From Alex, on Friday, September 13, 2019 11am:
  We weren't able to get this working on the final design, but Nikhil
  was, separately. I'm not sure if we were able to reconcile what was
  done differently, though.

  We shouldn't be skipping this step, as it's important for meeting
  timing constraints and fixing design rule errors. Doing so was more of
  a last ditch effort to get something out of Innovus. If I remember
  correctly, we thought optDesign wasn't finishing because congestion
  was too high at hotspots around the ICOVL cells. Did you look at the
  hotspot scores during placement?
```

steveri, 2019-09-20:
Update: Met with Mark, we decided that the problem has to do with congestion arising from ICOVL (litho alignment) cells, which form a vertical barrier in the middle of the chip, dividing the global buffer in two. The barrier is too wide for wires to cross w/o buffers, and standard cells / buffers are not allowed inside the barrier, so communication between right/left halves of global buffer gets pushed to the bottom, where congestion prevents finite-time routing.

The proposed solution is to open up horizontal paths above and below each alignment cell so wires can get through. It's not clear yet whether this is allowed by the design rules; it looks like maybe it would be okay, but changing the barrier will involve editing the ICOVL standard cells and/or finding new ones that have less blockage above and below.

A better solution might be to re-architect the global buffer with the vertical barrier in mind somehow...?

alexcarsello, 2019-09-20:
It should be possible to create channels between the ICOVL cells without violating design rules, as long as there are blockages of appropriate size around the cells. You wouldn't actually have to edit the ICOVL cells, just add more space between them and adjust the blockages we place over/around them.

I'll be fully back next Thursday 9/26, so I'll be able to help more with this effort then.

steveri, 2019-10-07:
Update: after the above comment, we did add space around each of the center-core icovl cells, and moved the entire array up a bit to alleviate particular congestion at the bottom of the global buffer. This helped things a bit and did reduce congestion but did not solve the problem of optDesign not completing.

What *did* solve the problem, as of this past weekend, was the addition of two directives, one to skip drc checking when (re)adding filler cells during optDesign, and one to skip eco routing during optDesign:
```
          setFillerMode -add_fillers_with_drc false -check_signal_drc false
          optDesign -postRoute -hold -setup -noEcoRoute
```
(Note `add_fillers_with_drc` was already false, but `check_signal_drc` was true.)

steveri, 2019-10-07:
I may continue tuning / refining optDesign, but routing no longer takes forever (it takes 7 hours or so now), so I'm going to go ahead and close this issue.

closed 2019-10-07 by steveri

steveri, 2019-10-09:
FYI I tried optDesign with and  optDesign with and without "`-noEcoRoute`" option.
With `-noEcoRoute`: completes in 7-8 hours
Without `-noEcoRoute`: still running after 15 hours, not clear if it's converging or not.
Also, we do ecoroute as a separate pass after optdesign anyway.
So it looks like, for finite optdesign runtime, we keep both "`-check_signal_drc false`" *and* "`-noEcoRoute`"