StanfordAHA/garnet #356: Tapeout: FE_FILLER cells and shorted nets

steveri, 2019-09-24:
The first set of errors that arise during full tapeout are related to shorted nets and FE_FILLER cells (see e.g. https://buildkite.com/tapeout-aha/top/builds/272  log "PNR(plan)"
```
[09/23 18:55:24 35730s] **ERROR: (IMPFP-4023): Could not assign any net to pin 'VDD' 
of instance 'FE_FILLER_N_1053', Because of short of nets 'VDD' and 'VSS'.
```
This appears to happen during the `do_plan` command in `floorplan.tcl`. `do_plan` is a proc defined in `vlsi/flow/scripts/gen_floorplan.tcl` where we see
```
    # [stevo]: add -logic so fillers get RTE signal connection
    add_io_fillers -cells "$ioFillerCells"   -derive_connectivity  -logic
```
The cells in question (e.g. `FE_FILLER_N_1053`) seem to adjoin analog IO pads `ANAIO*`. Which leads me to think this is all related to a warning earlier in `innovus.logv`
```
[09/23 11:02:54 130s] **WARN: (IMPDB-2078): Output pin AIO of instance
ANAIOPAD_ext_Vcal is connected to power net ext_Vcal.  Usually it is
not right to connect an output signal pin to a P/G net, unless the pin
is meant to be the driver of the net. This can create a short circuit
if the output is ground.  Check the connectivity in the netlist.
````
This is as far as I've gotten so far on this. Anyone who has insight, please chime in!

steveri, 2019-09-24:
There are a lot of warnings related to the buttertop_phy cell e.g.
```
Pin 'ext_Vcal' of cell 'butterphy_top' is declared as power/ground in
    LEF but as signal in timing library.  Treat it as power/ground.
**WARN: (IMPVL-159): Pin 'DVDD' of cell 'butterphy_top' is defined in LEF but not in the timing library.
**WARN: (IMPVL-159): Pin 'DVSS' of cell 'butterphy_top' is defined in LEF but not in the timing library.
**WARN: (IMPVL-159): Pin 'AVDD' of cell 'butterphy_top' is defined in LEF but not in the timing library.
**WARN: (IMPVL-159): Pin 'AVSS' of cell 'butterphy_top' is defined in LEF but not in the timing library.
**WARN: (IMPVL-159): Pin 'CVSS' of cell 'butterphy_top' is defined in LEF but not in the timing library.
**WARN: (IMPVL-159): Pin 'CVDD' of cell 'butterphy_top' is defined in LEF but not in the timing library.
```

Maybe I have the wrong butterphy cells? It's loading them from here: 
`/sim/ajcars/aha-arm-soc-june-2019/components/butterphy/butterphy_top.lib`

steveri, 2019-09-24:
`floorplan.tcl` reads an io file e.g. `read_io_file ../../../pad_frame/io_file  -no_die_size_adjust `

In my directory this points to an io_file in the garnet repo`"garnet/pad_frame/io_file"` ...  this file seems to lack certain items, e.g. these declarations for ANAIOPAD that I found in a competing io_file `"/sim/ajcars/to_nikhil/updated_scripts/io_file"`
```
    (inst name="ANAIOPAD_ext_clk_async_p" offset=1360)
    (inst name="ANAIOPAD_ext_clk_async_n")
    (inst name="ANAIOPAD_ext_clkn")
    (inst name="ANAIOPAD_ext_clkp")
    (inst name="ANAIOPAD_CVDD")
    (inst name="ANAIOPAD_CVSS")
```

My questions:
- where does io_file come from / who generates it?
- which one should I use
- etc.

I think I will try using the to_nikhil file and see if my errors go away...

steveri, 2019-09-25:
I tried out the new/different io_file (see prev comment). Looking at before-and-after logs
`/sim/steveri/garnet/tapeout_16/synth/gpf0_gold/innovus.log` and `/sim/steveri/garnet/tapeout_16/synth/gpf2_nikhil_padframe/innovus.log` ... it looks like this didn't make much difference in terms of solving the "shorted nets" problem
```
gpf0_gold/innovus.log:
 Couldnt assign pin 'RTE' of 'FE_FILLER_N_1119', nets 'rte' / 'rte_3' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_1207', nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_1229', nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_E_429',  nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_E_432',  nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_E_436',  nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_E_439',  nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'RTE' of 'FE_FILLER_E_465',  nets 'rte' / 'rte_3' shorted
 Couldnt assign pin 'RTE' of 'FE_FILLER_E_468',  nets 'rte_3' / 'rte' shorted
 Couldnt assign pin 'RTE' of 'FE_FILLER_E_477',  nets 'rte' / 'rte_3' shorted
 Couldnt assign pin 'RTE' of 'FE_FILLER_E_481',  nets 'rte_3' / 'rte' shorted

gpf2_nikhil_padframe/innovus.log:
 Couldnt assign pin 'RTE' of 'FE_FILLER_N_690',  nets 'rte_3' / 'rte' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_826',  nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_863',  nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_899',  nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_935',  nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_1009', nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_1046', nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_1083', nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'VDD' of 'FE_FILLER_N_1115', nets 'VDD' / 'VSS' shorted
 Couldnt assign pin 'RTE' of 'FE_FILLER_N_1479', nets 'rte' / 'rte_3' shorted
```

HOWEVER it does seem to have significantly reduced some other errors (see below). So I'm keeping the hack for now anyway.

Result of using new/different io_file:
- one less "Could not assign" error (10 vs. 11)
- eliminated this error(s)
'**ERROR: (IMPFP-7251):  block/IO pad/stand Cell not on Grid error is repeated over 5 times.'
- eliminated these 7 errors (four times) (once for each db read maybe)
```
**ERROR: (IMPFP-3356):  IO Inst IOPAD_top_VDD_dom3 is outside of design boundary...
**ERROR: (IMPFP-3356):  IO Inst IOPAD_top_VDDPST_dom3 is outside of design boundary...
**ERROR: (IMPFP-3356):  IO Inst IOPAD_bottom_POC_DIG is outside of design boundary...
**ERROR: (IMPFP-3356):  IO Inst ANAIOPAD_CVSS is outside of design boundary...
**ERROR: (IMPFP-3356):  IO Inst ANAIOPAD_CVDD is outside of design boundary...
**ERROR: (IMPFP-3356):  IO Inst ANAIOPAD_AVSS is outside of design boundary...
**ERROR: (IMPFP-3356):  IO Inst ANAIOPAD_AVDD is outside of design boundary...
Type 'man IMPFP-3356' for more detail.
```

Meanwhile the shorted-nets problem persists. I found something funny in gen_bumps, I will update in a separate comment...

steveri, 2019-09-25:
**Weirdness in gen_bumps, connects power to ground!!?** - Looking at gen_bumps procedure in gen_floorplan.tcl, I see this (below). It looks to me like this would short the power and ground nets. I.e. first command seems to connect VSS to VDDPST and the next connects VSS to VDD.

```
[09/24 14:06:43 160s] assign_bumps -multi_bumps_to_multi_pads -selected -pg_only 
-pg_nets VSS -pg_insts ${io_root}*VDDPST_* -exclude_region {1050 1050 3840 3840}

??? ASSIGNING VDD TO VSS ???
[09/24 14:06:43 160s] assign_bumps -multi_bumps_to_multi_pads -selected -pg_only 
-pg_nets VSS -pg_insts ${io_root}*VDD_* -exclude_region {1050 1050 3840 3840}
```
!!? Are we shorting the power and ground nets???

alexcarsello, 2019-09-25:
No, we are not shorting power and ground. The power pads in TSMC16 are power/ground "combo" pads, meaning that each of those pads has both a VDD and a VSS port that needs to be connected to a VDD and a VSS bump, respectively. Each of the power pad instances is named IOPAD_VDD<something here>.

steveri, 2019-10-01:
> No, we are not shorting power and ground.

Yes of course. Made perfect sense once I thought about it for two seconds. Duh :(
Meanwhile thanks @alexcarsello for pointing out that there really is no short, it's just that the filler cells' pins are mislabeled sorta. However, the tools still (erroneously) flag the error so we should find a fix. It will probably involve finding different filler cells with different labels. I'm keeing the issue open with the understanding that it is now much lower priority...

steveri, 2019-10-07:
Looks like it's not the filler cells that are mislabeled, so much as the analog IO cells, see enclosed picture. I think the tools will be happier if e.g. the AIO bottom stripe, the one that connects to VDD, were labeled VDD instead of VSS etc.  I will try and find analog IO cells that match better...

[image]

steveri, 2019-11-21:
Okay I finally found the problem, and/or at least I found a way around it. The original problem was this command in `vlsi/flow/scripts/gen_floorplan.tcl`.

    # [stevo]: add -logic so fillers get RTE signal connection
    add_io_fillers -cells "$ioFillerCells" -derive_connectivity  -logic

I found by trial and error that if I leave off the "-derive_connectivity" argument, the errors don't happen. I.e.
```
  good: add_io_fillers -cells "$ioFillerCells"
  good: add_io_fillers -cells "$ioFillerCells" -logic
  bad:   add_io_fillers -cells "$ioFillerCells" -logic -derive_connectivity
```
I sent e-mail to wiser heads to see if anyone sees a problem with this fix (below) ...pending responses we can maybe close this issue.
```
    # [stevo]: add -logic so fillers get RTE signal connection
    # add_io_fillers -cells "$ioFillerCells" -logic -derive_connectivity

    # [steveri 11/2019]: Dunno where "-derive_connectivity" comes from, but it
    # throws errors when used in conjunction w/Soong-jin's ANAIOPAD shenanigans
    add_io_fillers -cells "$ioFillerCells" -logic
```

steveri, 2019-11-22:
Yes the errors are gone, see below/after buildkite snapshots below. There are new/different errors (see below), but we'll leave that for another issue.

BEFORE - https://buildkite.com/tapeout-aha/top/builds/383
```
PNR(Plan) log:
  PNR SUMMARY
    grep ERROR innovus.log
    **ERROR: (IMPFP-4023): Could not assign any net to the pin 'VDD' of instance  'FE_FILLER_N_1047', Because of short of nets 'VDD' and 'VSS'.
    **ERROR: (IMPFP-4023): Could not assign any net to the pin 'RTE' of instance  'FE_FILLER_N_1113', Because of short of nets 'rte' and 'rte_3'.
    **ERROR: (IMPFP-4023): Could not assign any net to the pin 'VDD' of instance  'FE_FILLER_N_1201', Because of short of nets 'VDD' and 'VSS'.
...
```

AFTER - https://buildkite.com/tapeout-aha/top/builds/385
```
PNR(Plan) log:
  PNR SUMMARY
    grep ERROR innovus.log
    **ERROR: (IMPFP-7251): block/IO pad/stand Cell not on Grid
    **ERROR: (IMPFP-3967): Module Constraint is not on manufacturing grid
```

closed 2019-11-22 by steveri