StanfordAHA/garnet #367: Tapeout: LVS errors in final gds

steveri, 2019-10-14:
On r7arm-aha machine, see results in e.g. `/sim/steveri/tmpdir/lvs_top`
* results summary in `./GarnetSOC_pad_frame/GarnetSOC_pad_frame.lvs.report`
*  View results : `calibre -rve -lvs ./GarnetSOC_pad_frame/svdb`

The actual error relates to 6 random bits in a 20-bit register (??):
```
  % grep -i err /sim/steveri/tmpdir/lvs_top/GarnetSOC_pad_frame/GarnetSOC_pad_frame.lvs.report
  LVS Netlist Compiler - Errors and Warnings for "GarnetSOC_pad_frame.sp"
  Error: No matching pin name "RC_CG_GCLK_PORT" in subcircuit
    "Register_has_ce_True_has_reset_False_has_async_reset_False_type_Bits_n_16_4"
    in file "../pe/Tile_PE/Tile_PE.sp" at line 26106
  Also:
    "Register_has_ce_True_has_reset_False_has_async_reset_False_type_Bits_n_16_5"
    "Register_has_ce_True_has_reset_False_has_async_reset_False_type_Bits_n_16_9"
    "Register_has_ce_True_has_reset_False_has_async_reset_False_type_Bits_n_16_14"
    "Register_has_ce_True_has_reset_False_has_async_reset_False_type_Bits_n_16_16"
    "Register_has_ce_True_has_reset_False_has_async_reset_False_type_Bits_n_16_18"
```
You should be able to duplicate the errors by following the instructions in
  /sim/steveri/garnet/tapeout_16/lvs/README.txt

```
# Must be on arm machine
ssh r7arm-aha

# Set up for to run calibre
module load icadv/12.30.712
module load calibre/2019.1
# etc. also see ~ajcars/.cshrc

# Find a place to work
mkdir /sim/$USER/tmpdir; cd /sim/$USER/tmpdir

# top
cd /sim/$USER/tmpdir
mkdir lvs_top; cd lvs_top
lvs=/sim/steveri/garnet/tapeout_16/lvs/lvs.sh
top=GarnetSOC_pad_frame
gds=/sim/steveri/garnet/tapeout_16/synth/ref/final_final.gds
vlg=/sim/steveri/garnet/tapeout_16/synth/ref/pnr.final.lvs.v
ls -l $gds $vlg
echo $lvs $gds $vlg $top
$lvs $gds $vlg $top |& tee lvs_top.log
```

THofstee, 2019-10-14:
Are you synthesizing both tiles with power domains enabled?

steveri, 2019-10-14:
@THofstee 
> Are you synthesizing both tiles with power domains enabled?

Yes I think so. Should they be off?

THofstee, 2019-10-14:
It should be synthesizing with power domains enabled. I'm just wondering because the missing port looks like something related to clock gating.

I have one other idea, I was running into issues with PrimeTime before because the Tile_MemCore and Tile_PE both define the same modules with different implementations/port signatures.

What does `Register_has_ce_True_has_reset_False_has_async_reset_False_type_Bits_n_16_4` look like in Tile_PE's `pnr.sv` and Tile_MemCore's `pnr.sv`?

steveri, 2019-10-15:
Hm you may be onto something here. They do indeed appear to have same-name modules with different interfaces and definitions :( I'll have to look at this more closely when I get back from vacation (unless someone beats me to it...)

THofstee, 2019-10-15:
I looked at Genus a bit, I think we can work around this in the write_hdl command.

```
write_hdl
    [-suffix <string>]:
        specifies the string to be appended to the name of all defined modules
```

This would require the following changes:
- Tile_PE and Tile_MemCore each get a unique pre-determined suffix that needs to be added in their synthesis scripts
- Top-level synthesis probably needs to look for the tiles as `Tile_MemCore_suffix` instead of `Tile_MemCore`, and similarly for the PE.

steveri, 2019-10-15:
Nice! Thanks Teguh I will definitely try this when I get back.

steveri, 2019-11-06:
I think it worked, at least I'm not seeing the same errors as before. Unfortunately there is a new problem now, maybe caused by the fix...top-level LVS runs out of memory and crashes while flattening. I will close this issue and file a new one. :(

closed 2019-11-06 by steveri