StanfordAHA/garnet #375: Tapeout: Does IOPAD ordering matter?

steveri, 2019-11-13:
This actually pertains to the gitlab repo on the arm machine, but I'm filing the issue here with the rest of the tapeout issues.

In gitlab repo `aha-arm-soc-june-2019` there is a script `components/pad_frame/create_pad_frame.sh` for building the pad frame and associated io_file. There are also checked-in build results `example/io_file` and `./example/genesis_verif/GarnetSOC_pad_frame.sv`.

If you run `create_pad_frame.sh`, the resulting files `genesis_verif/GarnetSOC_pad_frame.sv` and `io_file` are the almost the same as the cached versions. But io_file is slightly different, in two ways
* ordering of the io pads
* presence/absence of `offset` keywords

This issue deals with the ordering of the IO pads. Specifically: 
Generated io_file pads are in a different order than the cached version `example/io_file`:

```
r7arm% cd aha-arm-soc-june-2019/components/pad_frame/
r7arm% ./create_pad_frame.sh
r7arm% cat io_file
...
(iopad
	(top
                ...
		(inst name="IOPAD_ext_dump_start" offset=3700)
		(inst name="IOPAD_top_VDD_dom3" offset=4500)
		(inst name="IOPAD_top_VDDPST_dom3" offset=4200)
		(inst name="IOPAD_top_rte_dom3")

% cat example/io_file
...
(iopad
	(top
                ...
		(inst name="IOPAD_ext_dump_start" offset=3700)
		(inst name="IOPAD_top_VDDPST_dom3" offset=4200)
		(inst name="IOPAD_top_VDD_dom3" offset=4500)
		(inst name="IOPAD_top_rte_dom3")
```

Questions:
* is order important?
* which one is correct??

ctorng, 2019-11-13:
The order doesn't matter here. What matters is which row/side the cell is placed on (top side) and the offset also matters (distance from the beginning of the row in counter-clockwise orientation).

So this puts `IOPAD_ext_dump_start` on the top row, at an offset 3700um from the top-left corner, then comes `IOPAD_top_VDDPST_dom3` (4200um), then `IOPAD_top_VDD_dom3` (4500um). So both snippets are the same.

```

                       3700um      4200um      4500um
    +---------+       +---------+ +---------+ +---------+
    |  corner | . . . |  IOPAD  | |  IOPAD  | |  IOPAD  | . . .
    +---------+       +---------+ +---------+ +---------+
                        ^           ^           ^
                        |           |           IOPAD_top_VDD_dom3
                        |          IOPAD_top_VDDPST_dom3
                       IOPAD_ext_dump_start
```

steveri, 2019-11-13:
Excellent! Thanks for the quick response. I will ignore the ordering discrepancy and close out this issue.

closed 2019-11-21 by steveri