StanfordAHA/garnet #376: Tapeout: Where do IOPAD offsets come from?

steveri, 2019-11-13:
This actually pertains to the gitlab repo on the arm machine, but I'm filing the issue here with the rest of the tapeout issues.

In gitlab repo `aha-arm-soc-june-2019` there is a script `components/pad_frame/create_pad_frame.sh` for building the pad frame and associated io_file. There are also checked-in build results `example/io_file` and `./example/genesis_verif/GarnetSOC_pad_frame.sv`.

If you run `create_pad_frame.sh`, the resulting files `genesis_verif/GarnetSOC_pad_frame.sv` and `io_file` are the almost the same as the cached versions. But io_file is slightly different, in two ways
* ordering of the io pads
* presence/absence of `offset` keywords

This issue deals with the `offset` keywords. Specifically: 
The original io_file has these extra "offset" keywords, which do not appear in the script-generated version:
```
r7arm% cd /sim/steveri/soc/components/pad_frame
r7arm% ./create_pad_frame.sh
r7arm% diff io_file example/io_file
7c7
<               (inst name="ANAIOPAD_ext_clk_async_p")
>               (inst name="ANAIOPAD_ext_clk_async_p" offset=1360)
17c17
<               (inst name="ANAIOPAD_ext_Vcm")
>               (inst name="ANAIOPAD_ext_Vcm" offset=2106)
24c24
<               (inst name="IOPAD_jtag_intf_i_phy_tck")
>               (inst name="IOPAD_jtag_intf_i_phy_tck" offset=3500)
30,32c30,32
<               (inst name="IOPAD_ext_dump_start")
<               (inst name="IOPAD_top_VDD_dom3")
<               (inst name="IOPAD_top_VDDPST_dom3")
---
>               (inst name="IOPAD_ext_dump_start" offset=3700)
>               (inst name="IOPAD_top_VDDPST_dom3" offset=4200)
>               (inst name="IOPAD_top_VDD_dom3" offset=4500)
```

For now I am adding a filter to the script to manually add the offsets. *This is not the right solution.*

My questions:
* why are the offsets there?
* what is the right solution for generating the offsets?

ctorng, 2019-11-13:
If offset isn't specified, I am pretty sure the cells are just abutted right next to each other (which may or may not be what you want).

Choosing pad placement is its own topic... E.g.:

- You want IO signals to come in close to where they are used
- You want to have enough IO power/ground pads (VDDPST/VSSPST) close to output signal pads to drive those signals out of the chip
- You want to spread core power (VDD/VSS) around the chip pretty evenly
- You want the clock signals to ideally come in a straight short path close to where the clock tree will start to fork

Ignoring all of that and just talking about the mechanics of placing the IO pads... in my experience, that IO file is pretty much the only way to do it, because (1) Innovus understands it, and (2) you have sufficient control over the details (offsets, spacing, indents). So you end up writing a script that generates that IO file (which I guess is what you did previously with the create_pad_frame.sh).

By the way:

```
              |---> this is `offset` counterclockwise from the corner

    +---------+       +---------+ +---------+ +---------+
    |  corner | . . . |  IOPAD  | |  IOPAD  | |  IOPAD  | . . .
    +---------+       +---------+ +---------+ +---------+
                     _            |--> this is `space`
                     |                 and is relative to
                     v                 the previous cell
          this way is `indent`
          for staggering the pads
          towards the chip center
```

ctorng, 2019-11-13:
The "right" offsets are usually chosen once, based on where you want IO signals to enter the chip and how you plan to package the chip, and then it shouldn't change much afterwards. So you don't really "regenerate" the offsets. The only thing that the script does is make it easier to express the Innovus IO file syntax and plug in the offsets you have already chosen.

Organization-wise, the IO pad frame is usually design-specific and I think we will end up with a similar approach to what you currently have.

steveri, 2019-11-13:
Okay. Thanks very much for the detailed explanation!

Meanwhile, in accordance with our "single source of truth" philosophy, I will eschew the awk filter in favor of embedding the offsets in the same file that Genesis uses to generates the pad_frame verilog (i.e. `create_pad_frame.sh` invokes Genesis, which reads `components/pad_frame/GarnetSOC_pad_frame.svp` and uses it to produce both `io_file` and `genesis_verif/GarnetSOC_pad_frame.sv`). I'm still interested in who/how/why these particular offsets were chosen for this particular pad frame, and/or if there is documentation behind that decision  (optimally would be part of the README in that directory maybe).

closed 2019-11-21 by steveri