StanfordAHA/garnet #377: What's up with cutmodule?

hofstee, 2019-11-14:
Why do we need to call cutmodule in the PD scripts? @steveri @alexcarsello

alexcarsello, 2019-11-14:
These are there because our flow is hierarchical. We are ensuring that references to our hierarchical blocks (Tile_PE and Tile_MemCore) are resolved using the hard macros we've already produced, not re-synthesized from the input verilog.

hofstee, 2019-11-14:
My weird problems should be fixed by #378 

New version of CoreIR doesn't decorate `endmodule` with the module's name anymore, so cutmodule as written would cut everything starting from the desired module until the end of the file is reached.

I think in SystemVerilog you might be allowed to have nested module declarations, so this, in theory, could break since it just looks for the first endmodule, but with the Verilog we're generating it should be fine.

steveri, 2019-11-15:
Thanks very much for finding and fixing this Teguh. I think the fix is good, and I finalized the pull, so I will close the issue.

closed 2019-11-15 by hofstee