StanfordAHA/garnet #382: Tapeout: RDL blockages?

steveri, 2019-12-02:
Context: In floorplanning, before routing bumps to pads, the proc `gen_route_bumps,` in `gen_floorplan.tcl,` calls a proc `gen_rdl_blockages.`

The resulting RDL blockages masked off most of the pad ring pins,
leaving only a narrow stripe. The stripe was not wide enough for the
RV-layer via that connected the rdl layer (AP) to the M9 pin on a
given pad, so most bumps remained unconnected at the end even if a
route was laid down. Meanwhile many bumps remained unrouted after
hours and hours of trying.

I commented out the "gen_rdl_blockages" and things seem much
happier. Instead of many hours (5-7 at least), multiple warnings and
poor result, the bump routing now takes just a few minutes and almost
(see separate issue) all bumps appear to be successfully connected to pads.

**My questions:**
- why did we have RDL blockages?
- how bad is it that I deleted them?

ctorng, 2019-12-02:
Has anything changed in the bump mappings since that file was put in? Maybe it was good for a previous mapping but not for this one?

I get the idea that those blockages were there primarily to force the automatic stuff to "do what you want".

I don't think it is bad to delete them or reduce how much they block.

steveri, 2019-12-04:
Thanks Chris, I agree and will close the issue.

closed 2019-12-04 by steveri