StanfordAHA/garnet #387: Tapeout: False(?) DRC violations on 45-degree wires - runlength spacing

steveri, 2019-12-18:
```
Special Wire of Net pad_tlx_rev_tdata_lo_p_i[21]
Actual:	0.900000	Required:	1.800000	
Type:	ParallelRunLength Spacing 
bbox = (4096.165, 2491.315) (4142.485, 2492.215)
```

45-degree routing happily produces wires that would end too close to a bump except for the fact that the end overlaps a manhattan wire such that the perceived gap does not really exist (see screenshot).

Unfortunately, later DRC checks flag this as an error. I believe this is incorrect, let me know if you disagree.  Also, let me know if you have a good way of preventing these false errors in the first place...

My plan is to flag these as false violations for now, so we can keep the existing flow. Later maybe Caliber will tell whether this is a real violation or not.

[image]

ctorng, 2019-12-18:
Yes, exactly.

This is a good motivation to just put the padframe through Calibre DRC. We won't know more about this until we get to Calibre, and we just need a GDS containing bump routing for this.

steveri, 2020-01-08:
I'm going to close this issue in favor of https://github.com/StanfordAHA/garnet/issues/389

For now we are doing manhattan routes and avoiding the problem.

closed 2020-01-08 by steveri