StanfordAHA/garnet #414: magma/peak dependency problems in requirements.txt

steveri, 2020-01-29:
When I check requirements.txt I get these two errors
```
  % cd $garnet; pip3 check requirements.txt
  magma-lang 2.0.6 has requirement ast_tools>=0.0.10, but you have ast-tools 0.0.3.
  peak 0.0.1 has requirement hwtypes>=1.3.0, but you have hwtypes 1.0.13.
```

and similar problems when I try to fix it by installing the packages via requirements.txt
```
  % pip3 install -r requirements.txt
  ERROR: magma-lang 2.0.6 has requirement ast_tools>=0.0.10,
    but you'll have ast-tools 0.0.3 which is incompatible.
  ERROR: peak 0.0.1 has requirement hwtypes>=1.3.0,
    but you'll have hwtypes 1.0.13 which is incompatible.
```

...not really sure what it's trying to say or what is the right way to fix it...?

Currently requirements.txt looks like this:
```
-e git://github.com/StanfordAHA/gemstone.git#egg=gemstone
-e git://github.com/StanfordAHA/canal.git#egg=canal
-e git://github.com/phanrahan/peak.git#egg=peak
-e git://github.com/StanfordAHA/lassen.git#egg=lassen
-e git://github.com/rdaly525/MetaMapper.git#egg=metamapper
-e git://github.com/Kuree/karst.git#egg=karst
-e git://github.com/joyliu37/BufferMapping#egg=buffer_mapping
-e git+git://github.com/pyhdi/pyverilog.git#egg=pyverilog
ordered_set
coreir>=2.0.50
magma-lang>=2.0.0
mantle>=2.0.0
cosa
-e git://github.com/leonardt/fault.git#egg=fault
hwtypes
archipelago
```

rsetaluri, 2020-01-29:
@steveri try `pip3 install -r requirements.txt --upgrade`? It looks like the actual required dependencies are *not* inconsistent but your currently installed ones are out of date.

steveri, 2020-01-29:
Thanks for the suggestions Raj.

I upgraded the two packages by hand and the requirements are now satisfied.
It seems like this is something that should happen automatically via requirements.txt, but maybe not.
I'll leave the issue open for awhile and if nobody has better ideas I'll close it.

leonardt, 2020-01-29:
By default `pip` won't do the upgrade unless you pass the `--upgrade` flag.  The way to force this in requirements.txt would be to require a specific version number (or commit for git repos)