StanfordAHA/garnet #417: coreir/pycoreir requirements

steveri, 2020-01-31:
So I installed `pycoreir`, per Raj's suggestion on Wednesday, and my `libcoreir` problems went away. Thanks!

Currently garnet `requirements.txt` includes a requirement for `coreir (coreir>=2.0.50)` but no mention of `pycoreir`. Should this change? https://github.com/StanfordAHA/garnet/blob/master/requirements.txt

For the record, these are the errors I got before and after deleting `libcoreir` files:
```
BEFORE deleting libcoreir*
AttributeError: /lib/libcoreir-c.so: undefined symbol: COREBitInOut

AFTER deleting libcoreir*
OSError: libcoreir-c.so: cannot open shared object file: No such file or directory
```

steveri, 2020-01-31:
New problem: everything seems fine when I compile default garnet e.g. `python garnet.py`. But if I request verilog output I get familiar problems again:
```
coreir: error while loading shared libraries: libcoreir.so: 
cannot open shared object file: No such file or directory
...
% python garnet.py -v
  File "/usr/local/lib/python3.7/site-packages/
  magma/backend/coreir_compiler.py", line 54, in __compile_verilog
    raise RuntimeError(f"CoreIR cmd '{cmd}' failed with code {ret}")

  RuntimeError: CoreIR cmd 'coreir -l corebit,coreir,global,float,\
  float_DW,commonlib -i garnet.json -o garnet.v' failed with code 127

```

I have both coreir and pycoreir installed atm, but of course no libcoreir as per prev fix.
```
  % pip3 list | grep coreir
  coreir         2.0.53   
  pycoreir       2.0.29.2 

  % ls /usr/lib/*coreir*
  <nothing>
```

How to fix? Am I missing something obvious here?

rdaly525, 2020-01-31:
Just to be clear 'coreir' in a python context refers to 'pycoreir'
'pip install coreir' should work fine. In general, do not manually remove coreirlib files...

rdaly525, 2020-01-31:
To fix your problem, I suspect the following will work:

```
pip3 uninstall coreir
pip3 install coreir
```

Also as good practice, you should almost always be in a virtual environment (like virtualenv or miniconda) when developing/testing these python tools

leonardt, 2020-01-31:
What is the path of `which coreir`? Is it `/usr/bin/coreir`?

You shouldn't have both `pycoreir` and `coreir` installed, I would just use the `coreir` package and uninstall `pycoreir`

leonardt, 2020-01-31:
(I'll see if I can remove the pycoreir package from pypi to prevent confusion)

leonardt, 2020-01-31:
CC @Kuree are you intending to maintain the `pycoreir` package? Right now it's very out of date.  If not, we should remove it from the `pypi` listing to prevent confusion.

leonardt, 2020-01-31:
(p.s. if you're using the `coreir` python pre-built package/binary, you should remove `/usr/bin/coreir` or `/usr/local/bin/coreir` from your path)

Kuree, 2020-01-31:
That was a hack to test coreir distribution. It's been deleted.

steveri, 2020-01-31:
Okay things seem to be working now. With help from Keyi, I essentially did the following:
```
rm /usr/lib/*coreir* /usr/bin/coreir /usr/local/bin/coreir
pip3 uninstall pycoreir
pip3 uninstall coreir
pip3 install coreir
```
Going forward I suppose other people won't have to worry about this particular problem; I had trouble because I had debris left over from previous installs. Maybe we should just close this issue with no further action(s) being taken?

rsetaluri, 2020-01-31:
Agreed @steveri; closing.

closed 2020-01-31 by rsetaluri