StanfordAHA/garnet #448: Long RTL generation times

alexcarsello, 2020-03-11:
Runtime for the 32x16 array of tiles (with the interconnect only option) has crept back up to ~25 minutes. It was around 5 minutes during the tapeout last year.

rsetaluri, 2020-03-11:
That's weird. @leonardt - what was the command you used to profile the python runtime last time?

leonardt, 2020-03-11:
I used cprofile: https://docs.python.org/3.7/library/profile.html

rsetaluri, 2020-03-11:
thanks @leonardt.

@alexcarsello can you run cprofile and paste the output here so we can see what's happening?

rsetaluri, 2020-03-17:
@alexcarsello how are we looking on this?

Kuree, 2020-05-06:
Just my own datapoint. On kiwi with update-to-date dependencies, 12x12 takes about 7 minutes to generate. Just for comparison, last year 12x12 takes about 1 minutes to generate.

rsetaluri, 2020-05-06:
To confirm you're running `python garnet.py --width 12 --height 12 --verilog` ?

I can c-profile and check that it's not anything silly if it is blocking.

Kuree, 2020-05-06:
Yes

leonardt, 2020-05-07:
@rsetaluri can you try https://github.com/phanrahan/magma/pull/701 and see if it helps?

rsetaluri, 2020-05-07:
@leonardt turns out a lot of the time discrepancy is due a gemstone-level issue (not caching properly).

Could be that further magma optimizations could help also. Will try this out, thanks.