StanfordAHA/garnet #761: PD: Innovus fails in QRC

steveri, 2021-03-12:
This happens fairly frequently, and is currently the most common cause of failure for a Friday Night Build.

The problem: Innovus crashes during an "optDesign" command (I guess, see error message below). When I restart the exact same job, it succeeds. This can happen in any step that runs QRC, e.g. `postroute_hold`.

As far as I can tell, optDesign performs RC extraction by way of a invoking a subtool called qrc(?), and it's this subtool that fails. The main error message seems to be:
```
  Cadence Quantus Extraction completed unsuccessfully at 2020-Aug-31 13:40:05
  (2020-Aug-31 20:40:05 GMT).
  "qrc" failed to finish successfully.

  **ERROR: (LS_PLACEOPT::runCurrentState): Internal failure executing '::LS_PLACEOPT::PR:doExtrInit'

  **ERROR: (IMPEXT-5016): Command qrc failed with error message: failed to run: 
  No such file or directory. Follow the reason in the error message to resolve the issue.
```

More complete error message and context below.
```
INFO (EXTGRMP-195) : Reading PINS section.
INFO (EXTGRMP-195) : Reading NETS section.
INFO (EXTGRMP-195) : Reading SPECIALNETS section.
ERROR (Cpp-2) : A mandatory condition failed to be true at line 907 of file snzcomps.cpp.
Condition: compFlags.to_bool()
ERROR (EXTGRMP-103) : Current job number 1 failed. Please check stdout and log files for more details. Exiting...
INFO (EXTHPY-254) : Extraction completed successfully at Mon Aug 31 13:40:02 2020.
Ending at 2020-Aug-31 13:40:05 (2020-Aug-31 20:40:05 GMT).

 Tool:                    Cadence Quantus Extraction 64-bit
 Version:                 19.1.1-s086 Mon Mar 25 09:39:10 PDT 2019
 IR Build No:             086 
 Techfile:                Unknown ; version: Unknown 
 License(s) used:         0 of Unknown 
 User Name:               buildkite-agent
 Host Name:               r7arm-aha
 Host OS Release:         Linux 3.10.0-862.11.6.el7.x86_64
 Host OS Version:         #1 SMP Tue Aug 14 21:49:04 UTC 2018
 Run duration:            00:00:00 CPU time, 00:01:34 clock time
 Max (Total) memory used: 0 MB
 Max (CPU) memory used:   0 MB
 Max Temp-Directory used: 0 MB
 Nets/hour:               0K nets/CPU-hr, 0K nets/clock-hr
 Design data:
    Components:           0
    Phy components:       0
    Nets:                 0
    Unconnected pins:     0
 Warning messages:        449
 Error messages:          2

Exit code 2.
Cadence Quantus Extraction completed unsuccessfully at 2020-Aug-31 13:40:05
(2020-Aug-31 20:40:05 GMT).
"qrc" failed to finish successfully.
**ERROR: (LS_PLACEOPT::runCurrentState): Internal failure executing '::LS_PLACEOPT::PR:doExtrInit' **ER
ROR: (IMPEXT-5016):      Command qrc failed with error message: failed to run: No such file or director
y. Follow the reason in the error message to resolve the issue.

 ReSet Options after AAE Based Opt flow 
Opt: RC extraction mode changed to 'detail'
cleaningup cpe interface
cleaningup cpe interface
*** Finished optDesign ***
Info: Destroy the CCOpt slew target map.
Removing temporary dont_use automatically set for cells with technology sites with no row.
**ERROR: (IMPSE-110):   File 'scripts/main.tcl' line 57: errors out.
### End verbose source output for 'scripts/main.tcl'.
**ERROR: (IMPSYT-6692): Invalid return code while executing 'START.tcl' was returned and script processing was stopped. Review the following error in 'START.tcl' then restart.
Error info: START.tcl: 
    while executing
"optDesign -postRoute -outDir reports -prefix postroute_hold -hold"
    (file "scripts/main.tcl" line 57)
    invoked from within
"::se_source_orig scripts/main.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel [concat ::se_source_orig $args]"
    (procedure "source" line 156)
    invoked from within
"source -verbose scripts/$tcl"
    ("foreach" body line 9)
    invoked from within
"foreach tcl $order {
  # Try to find the script in the "inputs" directory first
  if {[ file exists inputs/$tcl ]} {
    puts "\n  > Info: Sourcing \"..."
    (file "START.tcl" line 45)
    invoked from within
"::se_source_orig START.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel [concat ::se_source_orig $args]"
    (procedure "source" line 172)
    invoked from within
"source START.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel #0 source START.tcl"
    ("eval" body line 1)
    invoked from within
"eval {uplevel #0 source START.tcl}"
    (in namespace inscope "::" script line 1)
    invoked from within
"namespace inscope :: eval "uplevel #0 source $fileName"".
innovus 1>
```

ctorng, 2021-03-15:
This makes no sense to me... because it's non-deterministic.

If there was a way to replicate it in a smaller test case, we could probably dig into it. But if we cannot replicate it, then it is pretty difficult to dig into..

I am not sure exactly what to do here.

One early lead might be to just look at which module provides qrc ("which qrc"). And see if that module conflicts with other modules (common environment variables?).

Wasn't there a thing about Virtuoso / Calibre not playing together nicely if both are loaded at the same time? Hm..

steveri, 2021-03-15:
The right thing is probably of course to point our Cadence support guy at this issue ( `https://github.com/StanfordAHA/garnet/issues/761` ) and see if they recognize the error message(s) / info ... can someone do that for us, or at least remind me who the support person is, so I can do it?

ctorng, 2021-03-17:
We used to ask cloud support from Cadence for our VDE. But that was for issues on the VDE. I'm not sure who else to ask.

It would really help if we could somehow get a smaller case of this happening.

Another thought is to dump the output of the shell "env" command right before optDesign. Next time it fails, we will be able to see what is likely wrong. Specifically, the PATH variable will probably be different.

mahorowitz, 2021-03-17:
Just a wild guess.  Don't our systems dynamically load some file systems?  Is it possible that we are accessing a file that isn't mounted and thus sometimes fails.  On the retry it is present and runs?  Or do you think the error message is completely bogus.

steveri, 2021-03-17:
> Another thought is to dump the output of the shell "env" command right before optDesign. Next time it fails, we will be able to see what is likely wrong. Specifically, the PATH variable will probably be different.

It's worth a try...although the way our CI works, the environment really shouldn't change much from one run to the next...

steveri, 2021-03-17:
> Just a wild guess. Don't our systems dynamically load some file systems? Is it possible that we are accessing a file that isn't mounted and thus sometimes fails. On the retry it is present and runs? Or do you think the error message is completely bogus.

I think this is the likely thing. There is definitely weirdness associated with how the tool accesses its temporary space (see below). I generally find QRC files in three of the four named locations (`curdir`, `/tmp`, and `TMPDIR=/sim/tmp`); I'm thinking I should try setting `FE_TMPDIR` to "."

> **Specifying Temporary File Locations**
> You can specify a temporary file location for TQuantus and extraction. The temporary file location is chosen based on the following order of precedence:
>    1. If you specify a directory using the FE_TMPDIR environment variable, the software uses that directory as the temporary file location.
>    2. If you specify a directory using the TMPDIR environment variable, the software uses that directory as the temporary file location.
>    3. Saves the files to the current directory (if writable).
>    4. Saves the files to the /tmp directory.
>    
> Note: For IQuantus /TQuantus extraction in the distributed processing mode using different machines, do not store the cache or temporary data to the /tmp directory. _**The temporary data must be visible on all machines used for distributed processing. Either use the current directory, or specify a directory using the TMPDIR or FE_TMPDIR environment variable.**_
>

steveri, 2021-04-07:
Latest update, based on periodic testing:
* QRC seems to crash less often when multi-cpu is set to 8 rather than 16. Also relevant: user manual recommends not exceeding 8 because of diminishing returns (see below). I'll be filing a git pull soon to address this...

```
% xpdf /cad/cadence/INNOVUS19.10.000.lnx86/doc/innovusUG/innovusUG.pdf

April 2019 1057 Product Version 19.10, p. 1057

Distributed Processing

Multiple CPUs can be used to improve the overall turn-around time of
extraction. The run-time improvement may vary depending on multi-CPU
configuration, design size and type. Generally, performance
improvement will start to diminish beyond 8 CPUs.
```

steveri, 2021-05-27:
Fixed I guess, see mflowgen pull https://github.com/mflowgen/mflowgen/pull/104
and garnet pulls https://github.com/StanfordAHA/garnet/pull/777 , https://github.com/StanfordAHA/garnet/pull/783

closed 2021-05-27 by steveri