StanfordAHA/garnet #791: PD: Tile builds silently fail during final GLS step

steveri, 2021-06-09:
The step in question is the `pwr-aware-gls` step that happens when building a PE or memory tile. See below for a sample error log. The error has been going on for months, but because GLS fails silently and happens *after* a valid LEF has been produced, we have so far done nothing about it.

My plan is to "fix" mflowgen so that a failed GLS step will cause an actual error, at which point our build will break.

My questions:
* who owns this step and/or knows about it?
* given that it always fails, do we want to keep it as part of the build?
* if we do want to keep it, who is going to fix it and when?

Also see:
* GLS updates pull https://github.com/StanfordAHA/garnet/pull/743
* GLS fix pull https://github.com/StanfordAHA/garnet/pull/741
* GLS updates pull https://github.com/StanfordAHA/garnet/pull/290
* Pwr Aware Gate level simulations (GLS) pull https://github.com/StanfordAHA/garnet/pull/269

Here are the errors. Looks like an input port was not connected...

```
% less -N /build/gold.267/full_chip/19-tile_array/16-Tile_MemCore/28-pwr-aware-gls/mflowgen-run.log
      1 ++ irun -sv -timescale 1ns/1ns -access +rwc -notimingchecks -input cmd.tcl -ALLOWREDEFINITION inputs/adk/iocells-pwr.v inputs/adk/stdcells-lvt-pwr.v inputs/adk/s      1 tdcells-pm-pwr.v inputs/adk/stdcells-pwr.v inputs/adk/stdcells-ulvt-pwr.v -top tb_Tile_MemCore +define+TSMC_CM_NO_WARNING +define+TSMC_NO_TESTPINS_DEFAULT_VALUE_      1 CHECK inputs/design.vcs.pg.v tb_Tile_MemCore.v tb_Tile_PE.v
      2 TOOL:   xrun    19.03-s003: Started on Jun 08, 2021 at 10:07:30 PDT
      3 TOOL:   xrun(64)        19.03-s003: Started on Jun 08, 2021 at 10:07:30 PDT
      4 xrun(64): 19.03-s003: (c) Copyright 1995-2019 Cadence Design Systems, Inc.
      5 xrun: *W,NCEXDEP: Executable (irun) is deprecated. Use (xrun) instead.
      6 file: inputs/adk/iocells-pwr.v
      7         module worklib.PCBRTE_H:v
      8                 errors: 0, warnings: 0
      9         module worklib.PCBRTE_V:v
     10                 errors: 0, warnings: 0
..
   8468 file: tb_Tile_MemCore.v
   8469         module worklib.tb_Tile_MemCore:v
   8470                 errors: 0, warnings: 0
   8471 file: tb_Tile_PE.v
   8472         module worklib.tb_Tile_PE:v
   8473                 errors: 0, warnings: 0
   8474         Elaborating the design hierarchy:
   8475                 Caching library 'worklib' ....... Done
..
   8476    Tile_MemCore_RC_CG_MOD_3 RC_CG_HIER_INST3 (
   8477                                            |
   8478 xmelab: *W,CUVWSI (./inputs/design.vcs.pg.v,2414|43): 1 input port was not connected:
   8479 xmelab: (./inputs/design.vcs.pg.v,1833): test
..
   8480 
   8481    Tile_MemCore_RC_CG_MOD_4 RC_CG_HIER_INST4 (
   8482                                            |
   8483 xmelab: *W,CUVWSI (./inputs/design.vcs.pg.v,5591|43): 1 input port was not connected:
   8484 xmelab: (./inputs/design.vcs.pg.v,5044): test
..
   8485 
   8486    Tile_MemCore_RC_CG_MOD_5 RC_CG_HIER_INST5 (
   8487                                            |
   8488 xmelab: *W,CUVWSI (./inputs/design.vcs.pg.v,8374|43): 1 input port was not connected:
   8489 xmelab: (./inputs/design.vcs.pg.v,7778): test
..
   9756    Tile_MemCore_RC_CG_MOD_250 RC_CG_HIER_INST250 (
   9757                                                |
   9758 xmelab: *W,CUVWSI (./inputs/design.vcs.pg.v,207097|47): 1 input port was not connected:
   9759 xmelab: (./inputs/design.vcs.pg.v,207028): test
..
   9761 xrun: *E,ELBERR: Error during elaboration (status 1), exiting.
   9762 TOOL: xrun(64) 19.03-s003: Exiting on Jun 08, 2021 at 10:07:36 PDT
```

norabarlow, 2021-06-11:
I think Ankita would be the best person to look at this issue.

Is there another error hiding amongst the non-connected input ports? Usually not-connected ports aren't an issue, from my experience.

ankita0805, 2021-06-16:
I will check on this. Looks like the TB may need some updates.