StanfordAHA/garnet #900: PD: Cannot find RMUX cells during synthesis

steveri, 2022-08-08:
TSMC CI is currently failing in mem-tile synthesis, with the following error

https://buildkite.com/tapeout-aha/mflowgen/builds/5733
```
set rmux_cells [get_cells -hier RMUX_T*sel_inst0]
Warning : Could not find requested search value. [SDC-208] [get_cells]
        : The 'get_cells' command  cannot find any cells named 'RMUX_T*sel_inst0'

set_dont_touch $rmux_cells true
Error   : A required object parameter could not be found. [TUI-61] [parse_options]
        : An object of type 'subdesign|instance|net|design|libcell' named '' could not be found.
```

I know at least one way to fix this, and will submit a pull with something like (in `common.tcl`)
```
if { "$rmux_cells" != "" } {
    set rmux_cells [get_cells -hier RMUX_T*sel_inst0]
}
```

If someone reading this knows more about RMUX cells and why/whether there is a better fix, please let us know!

alexcarsello, 2022-08-08:
Don't do that... That's only kicking the problem down the road. We need to understand what changed because this constraint is important for breaking combinational loops.

steveri, 2022-08-08:
I have a potential fix that might satisfy your concerns...I believe this snippet of code works for both old and new RTL.

Instead of...
```
    set rmux_cells [get_cells -hier RMUX_T*sel_inst0]
```

...we do this...
```
    set rmux_cells [get_cells -hier RMUX_T*sel_*]
```
In the old RTL, this matches (only) old select instances `RMUX_T*sel_inst0` and in the new RTL it matches (only) new select instances `RMUX_T*sel_value_0`

alexcarsello, 2022-08-08:
Ok, that sounds good. Can you please make that change in all of the following places?

-PE synth constraints
-Mem synth constraints
-common/genlibdb-constraints
-common/synopsys-ptpx-gl/loop_break_Interconnect.tcl

steveri, 2022-08-08:
Please see new branch `missing-rmux-fix`, it is running regressions now, after which I will file a pull.
And yes, I came up with the same four files to change, so maybe we got it right :)

steveri, 2022-08-08:
The fix is in: see https://github.com/StanfordAHA/garnet/pull/901

steveri, 2022-08-09:
I don't think this worked :(

PE tile genlib step now takes 1.5 hours, whereas it used to take about 6 minutes.
(Ref. r7arm-aha:/build/pe5741/17-Tile_PE/25-cadence-genus-genlib)

The resulting lib file is almost 800 MB whereas it used to be less than 80 MB.

The new lib has 240K lines of the form
```
/* Start of combinational arc of pin SB_T4_WEST_SB_OUT_B16[14] wrt pin SB_T4_WEST_SB_IN_B16[0] */
/* Start of combinational arc of pin SB_T4_WEST_SB_OUT_B16[14] wrt pin SB_T4_WEST_SB_IN_B16[0] */
/* Start of combinational arc of pin SB_T4_WEST_SB_OUT_B16[14] wrt pin SB_T4_WEST_SB_IN_B16[0] */
...
```

Whreeas the old lib file had none of these.

I'm thinking we have to rejigger the following code in `genlibdb-constraints.tcl`
```
# These constraints ... [prevent] all downstream tools
# from analyzing combinational loops that can be realized
# in the interconnect.

set rmuxes [get_cells -hier *RMUX_*_sel_*]
set rmux_outputs [get_pins -of_objects $rmuxes -filter "direction==out"]
set_case_analysis 1 $rmux_outputs
```

If anyone already know the right thing to do, please speak up...meanwhile/otherwise I think I can track it down fairly quickly...

alexcarsello, 2022-08-09:
Yeah it didn't work if those paths are present. The RTL change must have been more than a simple name change.

steveri, 2022-08-09:
> Yeah it didn't work if those paths are present. The RTL change must have been more than a simple name change, or our constraints somehow didn't cover it properly.

I think maybe it really was a simple name change, but when I broadened the search pattern it brought in more cells than before. I.e. instead of `get_cells -hier *RMUX_*_sel_*` I should have done something more specific like `get_cells -hier -regexp .*RMUX_.*_sel_(inst0|value)` ... I am testing this now ...

alexcarsello, 2022-08-11:
@steveri do you have this fix ready to go?

steveri, 2022-08-11:
Yes, and it seems to be working quite well. Have a look if you like.
https://github.com/StanfordAHA/garnet/pull/904

I went with the more conservative pattern, it matches both old and new rmux names.

steveri, 2022-08-11:
I added this fix to the sparsity-mem branch and started a full-chip run, it seems to have finished the tile-array build in record time and is maybe on track to be our fastest TSMC full-chip build.
https://buildkite.com/tapeout-aha/fullchip/builds/422

steveri, 2022-08-17:
The fix seems to have worked, I am closing the issue now.

closed 2022-08-17 by steveri