StanfordAHA/garnet #1037: Magma-lang/mantle version

jiayi-wang98, 2024-02-05:
Hi,

I was trying to build this from scratch. And I'm getting this error "BUILD THE CGRA ERROR: __init__() got an unexpected keyword argument 'name_map'" when I tried to "python garnet.py --width 32 --height 16 --verilog".  

Currently, I'm using python==3.8, magma-lang==2.0.70, mantle==2.0.6.

Here's the shortened log:

--- GARNET-BUILD () 
....
-----------------------------------------------
--- Genesis Is Starting Work On Your Design ---
-----------------------------------------------
....

-----------------------------------------------
--- Genesis Finished Generating Your Design ---
-----------------------------------------------
Generating LALR tables
....
Ordt complete Sun Feb 04 23:55:42 PST 2024
- BUILD THE CGRA
Traceback (most recent call last):
  File "garnet.py", line 971, in <module>
    main()
  File "garnet.py", line 888, in main
    garnet = Garnet(args)
  File "garnet.py", line 67, in __init__
    self.interconnect = create_cgra(**cc_args.__dict__)
  File "/homes/iws/jwang710/aha_ws/garnet/cgra/util_onyx.py", line 490, in create_cgra
    core = PeakCore(pe_fc, ready_valid=ready_valid)
  File "/homes/iws/jwang710/aha_ws/garnet/peak_core/peak_core.py", line 135, in __init__
    self.wrapper = _PeakWrapper(peak_generator)
  File "/homes/iws/jwang710/aha_ws/garnet/peak_core/peak_core.py", line 36, in __call__
    self = super().__call__(peak_generator)
  File "/homes/iws/jwang710/aha_ws/garnet/peak_core/peak_core.py", line 52, in __init__
    circuit = peak_generator(MagmaFamily())
  File "/homes/iws/jwang710/aha_ws/garnet/src/peak/peak/features.py", line 141, in __call__
    res = self.fc(*args, **kwargs)
  File "/homes/iws/jwang710/aha_ws/garnet/src/lassen/lassen/sim.py", line 21, in PE_fc
    DataReg = gen_register_mode(DATAWIDTH, 0)(family)
  File "/homes/iws/jwang710/aha_ws/garnet/src/peak/peak/features.py", line 141, in __call__
    res = self.fc(*args, **kwargs)
  File "/homes/iws/jwang710/aha_ws/garnet/src/lassen/lassen/mode.py", line 30, in RegisterMode_fc
    Reg = family.gen_register(T, init)
  File "/homes/iws/jwang710/aha_ws/garnet/src/peak/peak/family.py", line 382, in gen_register
    return m.Register(T, init,
  File "/homes/iws/jwang710/python/lib/python3.8/site-packages/magma_lang-2.0.75-py3.8.egg/magma/generator.py", line 102, in __call__
    this = _make_type(cls, *args, **kwargs)
  File "/homes/iws/jwang710/python/lib/python3.8/site-packages/magma_lang-2.0.75-py3.8.egg/magma/generator.py", line 75, in _make_type
    cls.__init__(dummy, *args, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'name_map'.

I tried to use different versions of magma-lang and mantle but these error still occurs.

Could you kindly share which version of python/magma/mantle/... packages you are using and running for the current repo? I really appreciate any help.

Thank you,
Jiayi

steveri, 2024-02-05:
We generally use a docker container for all our builds. I don't suppose you have access to `stanfordaha/garnet:latest` on dockerhub?

jiayi-wang98, 2024-02-06:
> We generally use a docker container for all our builds. I don't suppose you have access to `stanfordaha/garnet:latest` on dockerhub?

No I don't have access to the dockerhub. Can I get that access some way? Otherwise, I guess I can only try on my local machine.

Anyway, thank you so much!

Best,
Jiayi

steveri, 2024-02-06:
I don't know how to get you on the dockerhub, but hopefully the information above will give you enough information to go forward on your local machine. Good luck!

jiayi-wang98, 2024-02-06:
Thank you so much. But it's still hard to run the example after trying to sync all the packages to the version mentioned. 

I have tried to modify the requirements.txt to the following:

coreir>=2.0.123  # should be first so we get the latest version
-e git+https://github.com/StanfordAHA/gemstone.git@072f1d67e6e6f0a654f429183e99b37c38f3eb08#egg=gemstone
-e git+https://github.com/StanfordAHA/canal.git#egg=canal
-e git+https://github.com/phanrahan/peak.git#egg=peak
-e git+https://github.com/StanfordAHA/lassen.git#egg=lassen
-e git+https://github.com/Kuree/karst.git#egg=karst
-e git+https://github.com/joyliu37/BufferMapping#egg=buffer_mapping
-e git+https://github.com/pyhdi/pyverilog.git#egg=pyverilog
-e git+https://github.com/StanfordAHA/lake@4625f4c96bef5e441600648af572e8a681fa658d#egg=lake-aha
-e git+https://github.com/phanrahan/magma.git@266aaee29d14238976e085ce69ef27f8aab3a81f#egg=magma-lang
-e git+https://github.com/phanrahan/mantle.git@8193cf6df60f5be7f198e55d74d9a196abec8342#egg=mantle
-e git+https://github.com/jack-melchert/peak_generator#egg=peak_gen
-e git+https://github.com/rdaly525/MetaMapper#egg=metamapper
ordered_set
cosa
-e git+https://github.com/leonardt/fault.git@3c3028f2fbaaa4ecf5527d6290a9ce5cc61be99a#egg=fault
hwtypes
-e git+https://github.com/Kuree/archipelago.git@8013be9085c42a5884495319667ffb7d0414be53#egg=archipelago
systemrdl-compiler
peakrdl-html
pulp

But I still get the error message below:
- BUILD THE CGRA
strg_ub_thin_PondTop
FOUND BULK
INPUTS DICT
{17: [{'strg_ub_thin_PondTop_flat': 'data_in_f_0'}, {'strg_ub_thin_PondTop_flat': 'data_in_f_1'}]}
Printing mode map...
{'strg_ub_thin_PondTop': (0, 'bulk')}
Printing mode map...
{'strg_ub_thin_PondTop': (0, 'bulk')}
Variable: strt_addr has no sink
Variable: strt_addr has no sink
Variable: strt_addr has no sink
Variable: strt_addr has no sink
Traceback (most recent call last):
  File "garnet.py", line 971, in <module>
    main()
  File "garnet.py", line 888, in main
    garnet = Garnet(args)
  File "garnet.py", line 67, in __init__
    self.interconnect = create_cgra(**cc_args.__dict__)
  File "/home/ubuntu/garnet/cgra/util_onyx.py", line 488, in create_cgra
    core = MemCore(use_sim_sram=use_sim_sram, gate_flush=not harden_flush, ready_valid=ready_valid)
  File "/home/ubuntu/garnet/memory_core/memory_core_magma.py", line 158, in __init__
    self.LT = LakeTop(data_width=self.data_width,
  File "/home/ubuntu/garnet/src/lake-aha/lake/top/lake_top.py", line 137, in __init__
    controllers.append(StrgUBVec(data_width=self.data_width,
  File "/home/ubuntu/garnet/src/lake-aha/lake/modules/strg_ub_vec.py", line 207, in __init__
    tb_only = StrgUBTBOnly(data_width=self.data_width,
  File "/home/ubuntu/garnet/src/lake-aha/lake/modules/tb_only.py", line 305, in __init__
    self.add_code(tb_to_out)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/kratos/generator.py", line 620, in add_always
    block_type, raw_sensitives, stmts = transform_stmt_block(self, fn, unroll_for=unroll_for,
  File "/home/ubuntu/.local/lib/python3.8/site-packages/kratos/pyast.py", line 1198, in transform_stmt_block
    src = astor.to_source(func_tree, pretty_source=__pretty_source)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/astor/code_gen.py", line 67, in to_source
    return pretty_source(generator.result)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/kratos/pyast.py", line 22, in __pretty_source
    return "".join(source)
TypeError: sequence item 13: expected str instance, list found

And it seems a problem with Kratos so I checked the version of kratos and tried to install it by 

pip install -e git+https://github.com/kuree/kratos@8c95d3073db27fd0d2cbe58197d07ec091a028bf#egg=kratos,
But still got errors for installing this kratos and tried multiple ways to manually build but all fails. 

I guess its hard to set up the environment without the docker so I will probably give up this. But anyway thank you so much!

steveri, 2024-02-07:
Here's one last thing you might try. I'm not sure what all these flags do, but this is what I use whenever I'm trying to build garnet RTL...
```
  $ flags="--width 4 --height 2 --pipeline_config_interval 8 -v --glb_tile_mem_size 256"
  $ flags+=" --interconnect-only"
  $ flags+=" --dump-config-reg"
  $ echo $flags | fold -sw 40  # optional
  $ python3 garnet.py $flags
```

jiayi-wang98, 2024-02-08:
Thank you so much although still not works. It would be really nice if you could send me the docker container file. I really appreciate it.

steveri, 2024-02-08:
I guess you can try and build the docker from source, just like we do. The source lives in our StanfordAHA/aha.git repo. So maybe you'd do something like this;
```
# Boot up a docker image

  git clone https://github.com/StanfordAHA/aha aha
  cd aha
  docker build . -t "garnet:my_image"
  docker pull garnet:my_image
  docker run -id --name my_container garnet:my_image bash
  source /aha/bin/activate

# (inside docker now) Assemble RTL-build flags for quick 4x2 build

  width=4
  height=$((width/2))
  flags="--width $width --height $height --pipeline_config_interval 8 -v --glb_tile_mem_size 256"

# To build older "amber" version of the chip (also see aha/bin/rtl-goldcheck.sh)

  export WHICH_SOC=amber
  # Update docker to match necessary amber environment
  garnet/mflowgen/common/rtl/gen_rtl.sh -u | tee tmp-amber-updates.sh
  cat tmp-amber-updates.sh
  source tmp-amber-updates.sh

# OR To build newer "onyx" version of the chip, do this instead

  export WHICH_SOC=onyx
  flags="$flags --rv --sparse-cgra --sparse-cgra-combined"

# Build the chip; result should be in 'garnet/garnet.v'

  aha garnet $flags

# Assemble final design.v
  cd garnet
  cp garnet.v genesis_verif/garnet.v
  cat genesis_verif/* > design.v
  cat global_buffer/systemRDL/output/glb_pio.sv >> design.v
  cat global_buffer/systemRDL/output/glb_jrdl_decode.sv >> design.v
  cat global_buffer/systemRDL/output/glb_jrdl_logic.sv >> design.v
  cat global_controller/systemRDL/output/*.sv >> design.v
```

jiayi-wang98, 2024-02-09:
Thank you so much for the instructions. I can almost run this through now. here's actually what I have done.

# Boot up a docker image

  git clone https://github.com/StanfordAHA/aha aha

  cd aha
++ git submodule init update --recursive

 sudo docker build . -t "garnet:my_image"
  -- docker pull garnet:my_image (not work without access, can still run through without this)
   sudo docker run -id --name my_container garnet:my_image bash
  source /aha/bin/activate (not work, use  "sudo docker exec -it aha_container bash" instead )

# (inside docker now) Assemble RTL-build flags for quick 4x2 build

  width=4
  height=$((width/2))
  flags="--width $width --height $height --pipeline_config_interval 8 -v --glb_tile_mem_size 256"

# To build older "amber" version of the chip (also see aha/bin/rtl-goldcheck.sh)

  export WHICH_SOC=amber
  # Update docker to match necessary amber environment
  garnet/mflowgen/common/rtl/gen_rtl.sh -u | tee tmp-amber-updates.sh
  cat tmp-amber-updates.sh
  source tmp-amber-updates.sh

# OR To build newer "onyx" version of the chip, do this instead

  export WHICH_SOC=onyx
  flags="$flags --rv --sparse-cgra --sparse-cgra-combined"

# Build the chip; result should be in 'garnet/garnet.v'

  aha garnet $flags (assert error if executing flags="$flags --rv --sparse-cgra --sparse-cgra-combined". Can run if not execute flags=...)

# Assemble final design.v
  cd garnet
  cp garnet.v genesis_verif/garnet.v
  cat genesis_verif/* > design.v
  cat global_buffer/systemRDL/output/glb_pio.sv >> design.v
  cat global_buffer/systemRDL/output/glb_jrdl_decode.sv >> design.v
  cat global_buffer/systemRDL/output/glb_jrdl_logic.sv >> design.v
  cat global_controller/systemRDL/output/*.sv >> design.v
  
  
  Thank you a ton for helping me this.

jiayi-wang98, 2024-02-09:
can run through with docker built.

closed 2024-02-09 by jiayi-wang98

steveri, 2024-02-09:
Oops @jiayi-wang98 they have just updated garnet master; the flags `—rv`, `—sparse-cgra` and  `—sparse-cgra-combined` have been eliminated. Anyone who uses the updated version will instead have to use `--include-sparse`. But if you can make it run successfully without using any of these flags, that's fine too.