StanfordAHA/garnet #1087: Verilator option for regressions?

steveri, 2024-10-04:
### THE CHALLENGE:
Can we provide verilator as an option for CGRA regression testing, as an alternative to existing vcs etc.

### EXECUTIVE SUMMARY of progress so far:
- It turns out that the verilog "clocking" construct, used in our existing testbench, is not supported by Verilator.
- My plan is to see if we can remove this construct from our testbench; opposing suggestions for a better plan are more than welcome!

### DETAILS
- Installed latest version of verilator
```
% verilator --version
Verilator 5.029 devel rev v5.028-160-gc05c48aaf
```
- Ran `pointwise` benchmark to build a populated testbench `/aha/garnet/tests/test_app`
- Built a bare-bones verilator testbench `CGRA.cpp`
- Tried it out
```
VFILES='
  /aha/garnet/global_buffer/header/global_buffer_param.svh
  /aha/garnet/global_buffer/header/glb.svh
  /aha/garnet/global_controller/header/glc.svh
  /aha/garnet/garnet.v
  /aha/garnet/global_buffer/systemRDL/output/glb_pio.sv
  /aha/garnet/global_buffer/systemRDL/output/glb_jrdl_decode.sv
  /aha/garnet/global_buffer/systemRDL/output/glb_jrdl_logic.sv
  /aha/garnet/global_controller/systemRDL/output/glc_pio.sv
  /aha/garnet/global_controller/systemRDL/output/glc_jrdl_decode.sv
  /aha/garnet/global_controller/systemRDL/output/glc_jrdl_logic.sv
  /aha/garnet/genesis_verif/*.sv
'
CW1=/cad/cadence/GENUS_19.10.000_lnx86/share/synth/lib/chipware/sim/verilog/CW/
CW2=/cad/cadence/GENUS_19.10.000_lnx86/share/synth/lib/chipware/sim/verilog/CWTECH/
verilator -Wno-fatal -Wall --cc CGRA.cpp --exe \
  /aha/garnet/global_buffer/header/global_buffer_param.svh \
  tb/top.sv \
  $VFILES \
  -F tb/tb_cgra.f \
  -y $CW1 -y $CW2 \
  libcgra.so
```
- Result:
```
    %Error-UNSUPPORTED: tb/axil_ifc.sv:102:20: Unsupported: Modport clocking
      102 |     modport driver(clocking cbd);
```

### OPTIONS GOING FORWARD
- fix verilator (not likely, see <https://github.com/verilator/verilator/issues/4555>)
- write and maintain a separate full-testbench for verilator
- rewrite existing testbench to not use clocking construct
- your name here?

### RESOURCES
- verilator "clocking" issue, see https://github.com/verilator/verilator/issues/4555
- <https://www.doulos.com/knowhow/systemverilog/systemverilog-tutorials/systemverilog-clocking-tutorial/>
- <https://www.chipverify.com/systemverilog/systemverilog-clocking-blocks>

### TODO
- I will make a garnet `verilator` branch containing my experiments including e.g. `CGRA.cpp`
- I will see what might be involved in rewriting our existing testbench without `clocking`

kalhankoul96, 2024-10-04:
Looks good Steve, let me know if you are able to figure it out.

steveri, 2024-10-08:
Here's a status update of how things are going so far...

#### Executive summary
I removed the clocking constructs and made a bunch of other changes (details below). And now the first phase of verilator simulation runs successfully.

#### Details
Verilator simulation occurs in three stages:
1. translate verilog source files into c++ equivalents
2. compile the c++ to make a simulator executable
3. run the simulator

Phase 1 now completes without error. Here's how that happened:

I removed all the `clocking` structures without regard to adjusting the timing, just to see if I could get Verilator to compile a simulator. This means that, if/when the simulator is running, it will probably get wrong answers until the timing is corrected again.

Old verilator did not understand timing delays, so I upgraded verilator to v5.

Gcc9 did not understand coroutines, so I upgraded gcc to gcc-10.

I turned off a BUNCH of warnings that will need to be revisited later

I made several changes w.r.t. directory `$GARNET/tests/test_app`:
* removed clocking signals from five testbench files
```
tests/test_app/tb/axil_driver.sv
tests/test_app/tb/axil_ifc.sv
tests/test_app/tb/environment.sv
tests/test_app/tb/proc_driver.sv
tests/test_app/tb/proc_ifc.sv
```
* rewrote non-compliant `$timeformat` calls in `tb/environment.sv`
* fixed(?) incorrect nonblocking assignments in `tb/top.sv` initial block
* made a local cache `vfiles` of all the verilog files needed for simulation
* made a reference script `bin/verilator.sh` for using vfiles etc. to build a verilator simulator
* everything is checked into the garnet repo 'verilator' branch as commit `eb9ad82be`; in theory, people can check out that commit and use `verilator.sh` to duplicate my progress so far (although they would possibly have to upgrade verilator and gcc first, see above)

#### What's next

* C++ compilation of the simulator has errors, these need to be squashed.

* Then we can try running the simulator to see what happens.

steveri, 2024-10-09:
Status update: Stage 2 works, verilator produces a simulator executable without error. Dunno yet if it actually works.

kalhankoul96, 2024-10-10:
Awesome! If the applications aren't working, let me know. I can help debug/look at waveforms.

steveri, 2024-10-10:
> Awesome! If the applications aren't working, let me know. I can help debug/look at waveforms.

Thanks! I hope to take you up on that soon. For now, no waveforms, the simulator is kind of a brick so far :(

kalhankoul96, 2024-10-14:
Great, awesome Steve. Let me know what help you need.

steveri, 2024-10-15:
Update:
* Simulation goes 210 ns without seg fault, believe it or not this is progress.
* It goes at a rate of about 2ns of sim time for every 1s of wall time. That's pretty terrible, but maybe it will be faster after the debug phase is over.
* This (below) is how far it gets in 210ns:
```
t=  100ns
Monitor initialization success
...
=====After Optimization=====
[ITER CTRL - loop 0] extent: 4094, cycle stride: 1, data stride: 2
Configuration of flush signal crossbar is updated to 0x0
[APP0-pointwise] group_start: 0
[APP0-pointwise] glb mapping success
Mapping kernel 0 Succeed
...
t=  119ns
Turn on interrupt enable registers
t=  210ns
- S i m u l a t i o n   R e p o r t: Verilator 5.028 2024-08-21
- Verilator: end at 210ns; walltime 100.779 s; speed 2.114 ns/s
- Verilator: cpu 99.561 s on 1 threads; alloced 512 MB
```

NEXT
* check everything in to verilator branch
* see what happens after 2000ns
* figure out how long it will take to do the whole thing maybe

steveri, 2024-10-23:
Progress report:

Someone (Kalhan I think) suggested that, to make Verilator work properly, we might have to replace all the object-oriented (class) code in our testbench with non-class-based equivalents. This seems to work, whereas other things I have tried have not worked, so I am going down that path. The class-based code comprises 1564 lines of verilog, while my translated equivalent (so far) totals about 500 lines, so maybe that means I'm about 1/3 of the way through. Continuing down this path means this translation phase should be done within a week or two, with any luck.

Also: I'm not sure why, but suddenly the verilated code is running much much faster than before. Earlier I reported 2ns of sim time for 1s of wall time, i.e. about 2 Hz; now it seems to be zipping through 500ns of simulation in well under a second, so, better than 500Hz at least...

steveri, 2024-11-06:
Finally tried running my in-progress verilator changes back on vcs, and things failed.

So then I went down a long rat-hole getting things back in whack again.

Now, the code compiles without error on both verilator and vcs platforms.

The vcs version gets correct answers, the verilator version not so much. On-going development aims to correct that.

Meanwhile, development has moved from the `verilator` branch to a new branch `vcs`.

steveri, 2024-11-12:
#### The good news
Had a little breakthrough yesterday. Verilator successfully runs pointwise, with cycle-accurate performance vs. the baseline VCS implementation, at least as far as I can tell.

#### The bad news.
Unfortunately, the two simulators are producing different results in `hw_output.txt.` :(
The Verilator run emits an extra four pixels at the beginning of the output file. Other than that, the two are identical:
```
            hw_output.txt.vcs       hw_output.txt.ver
            -------------------     -------------------
   0                             >  0150 0148 0144 0146
   1        0150 0148 0144 0146     0150 0148 0144 0146
   2        0146 0146 0148 014a     0146 0146 0148 014a
   3        0150 0152 0154 0150     0150 0152 0154 0150
 ...        ...                     ...
1023        0144 013a 0146 0150     0144 013a 0146 0150
1024        0152 0154 015a 0152     0152 0154 015a 0152
1025        0150 015c 0168 016c  <               
```

### What's next

Seems like a simple thing that I should be able to track down pretty quickly haha...

Meanwhile I tagged the latest commit locally as 'verilator-v0.4'
```
% git tag -n; git show-ref --tags
  a88cc6b  verilator-v0.0  VCS and Verilator both run without error
  57bfa53  verilator-v0.1  Code is running classlessly I think
  3222727  verilator-v0.2  Fixed (awready) no-reset-posedge bug
  a40fdfc  verilator-v0.3  Verilator (and VCS) PASSES pointwise
  7f969c6  verilator-v0.4  Verilator cycle-accurate vs. VCS on pointwise
```

steveri, 2024-11-13:
#### Executive summary
It works! Finally.

#### Details
Verilator successfully runs pointwise, with cycle-accurate performance vs. the baseline VCS implementation, at least as far as I can tell, and, very importantly, *it emits the correct output image* in `hw_output.txt`.

This is the procedure I use to verify correctness (on kiwi):

```
########################################################################
# Clone garnet repo into a dir $GARNET. Then:

cd $GARNET/tests/test_app/
git checkout verilator-v0.5   # <or if that doesn't work, maybe 'git checkout 9355720'>
mkdir vfiles

########################################################################
# Create or otherwise acquire garnet verilog and copy it to $GARNET/tests/test_app/vfiles/
<...>

########################################################################
# Create or otherwise acquire pointwise test collateral e.g. 
#   $GARNET/tests/test_app/test_data/pointwise/bin/hw_input_stencil.raw
#   $GARNET/tests/test_app/test_data/pointwise/bin/pointwise.bs
<...>

########################################################################
# Simulate using VCS, output to vcs.log
APP=+APP0=/nobackup/steveri/github/garnet/tests/test_app/test_data/pointwise
test -e hw_output.txt && rm hw_output.txt  // Remove results of prev run
vcs.sh
echo run | simv -lca -l simv.log +vcs+initmem+0 +vcs+initreg+0 -sv_lib libcgra -exitstatus -ucli $APP
mv hw_output.txt hw_output.txt.vcs

########################################################################
# Simulate using Verilator, output to ver.log
test -e obj_dir/ && /bin/rm -rf obj_dir/  # Remove previous build if needed
test -e hw_output.txt && rm hw_output.txt  // Remove results of prev run
verilator.sh
make -C obj_dir/ -f Vtop.mk
obj_dir/Vtop 8000 "$APP" |& tee vtop.log
cp hw_output.txt hw_output.txt.ver

########################################################################
# Verify that both VCS and Verilator got the same answer
diff simv.log vtop.log  # May not be exact but should be similar
diff hw_output.txt.vcs hw_output.txt.ver  # Should be exact match!

########################################################################
# Verify that both VCS and Verilator got the CORRECT answer
pw=$GARNET/tests/test_app/test_data/pointwise/bin
od -An -x --endian=big $pw/hw_output.raw | sed 's/^ //' > tmp.gold
diff -Bb hw_output.txt tmp.gold  # Should be NO diffs
```

#### Next steps
- slowly integrate the changes into the garnet main branch
- implement some kind of CI that can track when/whether the verilator version breaks
- ??

steveri, 2024-11-18:
Adding Yu-Chen so he can be in the loop...

steveri, 2025-02-17:
This works well enough for now, I am closing the issue.

closed 2025-02-17 by steveri